YosysHQ/sby #7: Multiple files per line in the files section

ZipCPU, 2018-04-13:
I just tried placing two files per line in the files section and was surprised that symbiyosys tried to copy the first file onto the second.  This created files in my filesystem outside of the SymbiYosys directory that I wasn't expecting.  Under some circumstances, this might even overwrite information (it didn't for me, but I can see how it might).

Request SymbiYosys use white space to delimit files in this section, rather than just lines, or at least detect when more than one item is on a line and issue an error.

cliffordwolf, 2018-04-13:
I've now added documentation of the `[files]` section syntax: http://symbiyosys.readthedocs.io/en/latest/reference.html#files-section

closed 2018-04-13 by cliffordwolf

cliffordwolf, 2018-04-13:
Added check for malformed dst filename in 836d54d4c7a0db9072026d81bed1ca91c3c913a3.