YosysHQ/sby #19: Invalid command async2sync added to yosys script

arjenroodselaar, 2018-11-04:
I am running the following SymbiYosys script, but encounter an error when yosys parses the resulting command file:

```
[options]
mode bmc
depth 20
append 2

[engines]
smtbmc

[script]
verilog_defaults -add -sv -formal
read_verilog -DALU_TOP hdl/alu.sv
prep -flatten -top alu
```

Error:
```
Building: finished in 0.0 sec (100%) 5/5 jobs, 0 updated
  Total time: 0.2 sec
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] Removing direcory '/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu'.
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] Writing '/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu/src/hdl/alu.sv'.
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] engine_0: smtbmc
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] base: starting process "cd /Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu/src; /Users/arjen/dev/snes-toolchain/bin/yosys -ql ../model/design.log ../model/design.ys"
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] base: ERROR: No such command: async2sync (type 'help' for a command overview)
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] base: finished (returncode=1)
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] base: job failed. ERROR.
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:23:41 [/Users/arjen/dev/risc-v/buck-out/gen/hdl/verification/alu.sby/alu] DONE (ERROR, rc=16)
```

Looking at syb_core.py:295, it seems that command is added because multiclock is off by default. Enabling multiclock makes the yosys script run to completion. I am running tip of master for both SymbiYosys and Yosys and could obviously hack around this, but I am not sure what is correct here.

Oh and thank you for these amazing tools!

daveshah1, 2018-11-04:
Are you sure your installed version of Yosys is up to date? It looks like async2sync was added back in July.

cliffordwolf, 2018-11-04:
What @daveshah1 said. Please update to latest Yosys.

closed 2018-11-04 by cliffordwolf