YosysHQ/sby #21: Unsupported cell type error caused by simple always block with 2 registers

tmeissner, 2018-11-30:
When I add a always block with 2 register assignments to my test bench code, I get an error about unsupported cell type during generation of SMT output.

The code looks like this:

```verilog
reg [15:0] last_opc;
reg [15:0] last_len;

  always @(posedge ClkSyst_i) begin
    if (HostIfValid_i && HostIfStart_i && !HostIfStop_i) begin
      last_opc <= HostIfData_i[15:4];
      last_len <= HostIfData_i[31:16];
    end
  end
```

The error I get looks like this:

```
...
SBY 15:42:37 [symbiyosys] base: finished (returncode=0)
SBY 15:42:37 [symbiyosys] smt2: starting process "cd symbiyosys/model; yosys -ql design_smt2.log design_smt2.ys"
SBY 15:42:37 [symbiyosys] smt2: -- (c) Copyright 1999 - 2018 Verific Design Automation Inc. All rights reserved
SBY 15:42:37 [symbiyosys] smt2: -- Compilation time was Tue Sep  4 22:13:47 2018
SBY 15:42:37 [symbiyosys] smt2: -- This program will expire on Thu Dec 13 21:13:47 2018
SBY 15:42:37 [symbiyosys] smt2: ERROR: Unsupported cell type $_DFFE_PP_ for cell HostRegisterFileF.$auto$simplemap.cc:442:simplemap_dffe$2980.
SBY 15:42:37 [symbiyosys] smt2: finished (returncode=1)
SBY 15:42:37 [symbiyosys] smt2: job failed. ERROR.
```

If I remove the assignment of one of the two registers, no error occurs.

The test bench looks liks this:
```systemverilog
module HostRegisterFileF (
  // global
  input        ClkSyst_i,
  input        ResetN_i,
  // VAI req
  input [63:0] HostIfData_i,
  input        HostIfStart_i,
  input        HostIfStop_i,
  input        HostIfValid_i,
  output       HostIfAccept_o,
  // VAI ack
  output [63:0] HostIfData_o,
  output        HostIfStart_o,
  output        HostIfStop_o,
  output        HostIfValid_o,
  input         HostIfAccept_i
);

  `define COUNTER_NUMBER 16

  HostRegisterFileE #(.G_COUNTER_NUMBER(`COUNTER_NUMBER)) i_HostRegisterFileE (
    .ClkSyst_i(ClkSyst_i),
    .ResetN_i(ResetN_i),
    .HostIfData_i(HostIfData_i),
    .HostIfStart_i(HostIfStart_i),
    .HostIfStop_i(HostIfStop_i),
    .HostIfValid_i(HostIfValid_i),
    .HostIfAccept_o(HostIfAccept_o),
    .HostIfData_o(HostIfData_o),
    .HostIfStart_o(HostIfStart_o),
    .HostIfStop_o(HostIfStop_o),
    .HostIfValid_o(HostIfValid_o),
    .HostIfAccept_i(HostIfAccept_i)
  );

  // Constrain reset

  reg init_flag = 1;
  (* gclk *) wire gbl_clk;

  always @(*) begin
    if (init_flag) assume (!ResetN_i);
    if (!init_flag) assume (ResetN_i);
  end

  always @(posedge ClkSyst_i)
    init_flag <= 0;

  global clocking
    @(posedge gbl_clk);
  endclocking

  always @($global_clock) begin
    assume (ClkSyst_i != $past(ClkSyst_i));
  end

  reg [15:4] last_opc;
  reg [15:0] last_len;

  always @(posedge ClkSyst_i) begin
    if (HostIfValid_i && HostIfStart_i && !HostIfStop_i) begin
      last_opc <= HostIfData_i[15:4];
      last_len <= HostIfData_i[31:16];
    end
  end

  assert property (@(posedge ClkSyst_i)
    disable iff (!ResetN_i)
      HostIfValid_o && HostIfStop_o && last_opc[11:4] == 8'b10 && last_len == 0 |->
      HostIfData_o[31:16] == 8
  );

  assert property (@(posedge ClkSyst_i)
    disable iff (!ResetN_i)
      HostIfValid_o && HostIfStop_o && last_opc[11:4] == 8'b11 && last_len == 0 |->
      HostIfData_o[31:16] == `COUNTER_NUMBER*8
  );

endmodule
```

My SymbiYosys script file:
```
[options]
mode bmc
#mode prove
depth 20
#multiclock on

[engines]
smtbmc
#abc pdr

[script]
verific -work BootFpga -vhdl DefinesP.vhd
verific -work Crc16    -vhdl Crc16MappingsP.vhd
verific -work Crc16    -vhdl Crc16P.vhd
verific -work Crc16    -vhdl Crc16Gen64E.vhd
verific -work work     -vhdl HostRegisterFileE.vhd
verific -sv HostRegisterFileF.sv
#prep -top HostRegisterFileF
hierarchy -check -top HostRegisterFileF
proc; opt
check
wreduce
alumacc
fsm
memory -nomap
memory_map
#techmap
dff2dffe
techmap -map +/adff2dff.v
hierarchy -check -top HostRegisterFileF
stat
check
synth -top HostRegisterFileF

[files]
HostRegisterFileE.vhd  
HostRegisterFileF.sv
```

I don't use the prep command because this results in another similar error, but with another FF type. With the dff2dffe & techmap -map +/adff2dff.v I was able to workaround that.

At the moment I can't share the DUT code, because is it confidential and I believe it isn't the source of this particular problem.

ZipCPU, 2018-12-01:
Please try again with this simplified yosys script:

    [script]
    verific -work BootFpga -vhdl DefinesP.vhd
    verific -work Crc16    -vhdl Crc16MappingsP.vhd
    verific -work Crc16    -vhdl Crc16P.vhd
    verific -work Crc16    -vhdl Crc16Gen64E.vhd
    verific -work work     -vhdl HostRegisterFileE.vhd
    verific -sv HostRegisterFileF.sv
    prep -top HostRegisterFileF

Yosys _should_ be able to fill in any additional requirements you need beyond the prep command.

I know you had said you used this before and it failed, but that other failure is really the failure I think we should be debugging, rather than the one posted above.

Dan

cliffordwolf, 2018-12-16:
Hi, what's your reason for running the `dff2dffe` command in your script?

tmeissner, 2018-12-16:
It’s because the RTL dut causes loop errors when using the prep command. With replacing the prep the error didn’t cause the verification process stop, but I got the error above. I assume it would be better to find the cause of the loops Yosys detected.

tmeissner, 2018-12-16:
Other synthesis tools like Synplify compile the design without errors. Unfortunately the design is confidential and to generate a simple test case without disclosing the design functionality for the error is hard 😟

ZipCPU, 2018-12-17:
Can you share at all how you are using $anyseq?  That may be the source of the logic loop you are struggling with.  Trying to reverse engineer what you've given, it looks like you are declaring certain values to have the $anyseq property, yet also giving those values explicit definition.

tmeissner, 2018-12-17:
@ZipCPU I don't use $anyseq. The only SV function I use is the (* gclk *) construct which you have suggested when using multi clock. I don't know exactly if it is necessary here. The DUT design is written in VHDL.

cliffordwolf, 2018-12-17:
> It’s because the RTL dut causes loop errors when using the prep command.

This explains why you are using a custom script instead of prep. It does not explain why you feel the need to include the `dff2dffe` command in your custom script. Why do you think you need that command?

tmeissner, 2018-12-18:
I tried the prep 'unrolling' which was suggested in a post at reddit. After that I got some errors which I tried to fix with the `dff2dffe` command, suggested by another reddit thread. It was more trial & error and not very methodic I admit.

I've chatted with @ZipCPU and he suggested that the `--syn` option of smtbmc engine could help. I will try that when I have a new eval verific license.

ZipCPU, 2018-12-19:
Just to give credit where credit is due ... it was actually @cliffordwolf who had the `--syn` suggestion that I then relayed to you ...

tmeissner, 2018-12-30:
With the -`-syn` option the logic loop warnings still exists but they don't cause an error anymore. So now the tool continues after the warnings. I will close this bug as the specific problem seems to be solved. Thanks for your help.

closed 2018-12-30 by tmeissner