YosysHQ/sby #23: initial assumptions do not work as expected

ZipCPU, 2018-12-19:
Consider the following cover test:
````
module tst(i_clk, i_reset, o_output);
    input wire i_clk, i_reset;
    output reg[2:0] o_output;

    initial o_output = 1;
    always @(posedge i_clk)
    if (i_reset)
        o_output <= 0;
    else
        o_output <= o_output + 1'b1;

    initial assume(i_reset);
    always @(*)
        cover(&o_output);
endmodule
````

# Actual Behavior

On the first clock, `i_reset` is false and `o_output` is one.  `o_output` increments from there, and `i_reset` is never raised.

# Expected Behavior

`i_reset` should be asserted on the first clock step, and `o_output` should then be zero on the next.

# Steps to reproduce

The symbiyosys script is fairly unremarkable: mode cover, engine smtbmc, read -formal tst.sv, prep -top tst, etc.

Dan

tudortimi, 2020-04-12:
Using `initial` like this would be great, as it would be much shorter than the current idiom:

```
bit first_cycle = 1;
always @(posedge i_clk)
  first_cycle <= 0;

always @(posedge i_clk)
  if (first_cycle)
    assume(i_reset);
```

tudortimi, 2020-04-12:
I was going to propose generalizing the `first_cycle` concept somehow with a custom system function, `$first_cycle()`, but `initial` is much nicer.

ZipCPU, 2020-04-15:
Unfortunately, Verific doesn't want to support our need for this from `initial` statements.  Our plan is currently to implement it on our own, but it's not as simple to do as one might expect.  In the meantime, you can also use the `chformal` command to skip evaluation of the first time step.  This isn't a perfect solution either, since combinatorial checks and sequential checks report on different cycles, so skipping one cycle might help for combinatorial checks but not sequential ones, and skipping two cycles will leave combinatorial checks unchecked for a cycle.

Dan

tudortimi, 2020-04-17:
I stumbled across `$initstate`, which does what I described for `$first_cycle`.