YosysHQ/sby #24: Link cover statement to a particular trace

tomverbeure, 2018-12-31:
When you have many cover statements in a single file, you get the same amount of cover traces.

Is there a way to identify which cover statement is linked to which trace?

I'm looking for something that might look like this:
```
   my_cover: cover(....)
```
This would result in a `trace_my_cover.vcd`.

cliffordwolf, 2019-01-02:
The problem with this request is that there is no 1:1 relationship between cover statement (and assertions) and the traces. Instead, one trace may cover more than one cover statement, or violate more than one assertion. The logfile output of SymbiYosys contains the information which statements are covered/violated by a given trace. For example:

```
SBY 16:42:52 [cover] engine_0: ##   0:00:00  Solver: boolector
SBY 16:42:53 [cover] engine_0: ##   0:00:00  Checking cover reachability in step 0..
SBY 16:42:53 [cover] engine_0: ##   0:00:00  Checking cover reachability in step 1..
SBY 16:42:53 [cover] engine_0: ##   0:00:00  Checking cover reachability in step 2..
SBY 16:42:53 [cover] engine_0: ##   0:00:01  Checking cover reachability in step 3..
SBY 16:42:54 [cover] engine_0: ##   0:00:01  Checking cover reachability in step 4..
SBY 16:42:54 [cover] engine_0: ##   0:00:01  Checking cover reachability in step 5..
SBY 16:42:54 [cover] engine_0: ##   0:00:02  Checking cover reachability in step 6..
SBY 16:42:55 [cover] engine_0: ##   0:00:02  Checking cover reachability in step 7..
SBY 16:42:56 [cover] engine_0: ##   0:00:03  Checking cover reachability in step 8..
SBY 16:42:56 [cover] engine_0: ##   0:00:03  Checking cover reachability in step 9..
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement at cover.sv:65 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement at cover.sv:66 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement at cover.sv:64 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement at cover.sv:61 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Writing trace to VCD file: engine_0/trace0.vcd
SBY 16:43:05 [cover] engine_0: ##   0:00:12  Writing trace to Verilog testbench: engine_0/trace0_tb.v
SBY 16:43:05 [cover] engine_0: ##   0:00:12  Writing trace to constraints file: engine_0/trace0.smtc
SBY 16:43:05 [cover] engine_0: ##   0:00:12  Checking cover reachability in step 9..
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Reached cover statement at cover.sv:67 in step 9.
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Reached cover statement at cover.sv:62 in step 9.
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Reached cover statement at cover.sv:68 in step 9.
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Writing trace to VCD file: engine_0/trace1.vcd
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Writing trace to Verilog testbench: engine_0/trace1_tb.v
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Writing trace to constraints file: engine_0/trace1.smtc
SBY 16:43:06 [cover] engine_0: ##   0:00:13  Checking cover reachability in step 9..
SBY 16:43:06 [cover] engine_0: ##   0:00:14  Checking cover reachability in step 10..
SBY 16:43:07 [cover] engine_0: ##   0:00:14  Checking cover reachability in step 11..
SBY 16:43:11 [cover] engine_0: ##   0:00:18  Reached cover statement at cover.sv:60 in step 11.
SBY 16:43:11 [cover] engine_0: ##   0:00:19  Writing trace to VCD file: engine_0/trace2.vcd
SBY 16:43:12 [cover] engine_0: ##   0:00:19  Writing trace to Verilog testbench: engine_0/trace2_tb.v
SBY 16:43:12 [cover] engine_0: ##   0:00:19  Writing trace to constraints file: engine_0/trace2.smtc
SBY 16:43:12 [cover] engine_0: ##   0:00:19  Checking cover reachability in step 11..
SBY 16:43:12 [cover] engine_0: ##   0:00:19  Reached cover statement at cover.sv:59 in step 11.
SBY 16:43:12 [cover] engine_0: ##   0:00:19  Writing trace to VCD file: engine_0/trace3.vcd
...
```

Re labels on `assert` or `cover` statements: Yes, we definitely need to support those.

tmeissner, 2019-01-03:
Okay, sounds reasonable. Would it then be possible to have the assert/cover labels in the log in addition to the line numbers? Maybe someone (like me ;)) wants create a script which parses the SymbiYosys output and generates a coverage report from that.

Something likes this:

```
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement cover_read_ack at cover.sv:65 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement cover_write_ack at cover.sv:66 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement cover_write_req at cover.sv:64 in step 9.
SBY 16:43:04 [cover] engine_0: ##   0:00:12  Reached cover statement cover_read_req at cover.sv:61 in step 9.
```

cliffordwolf, 2019-01-03:
> Would it then be possible to have the assert/cover labels in the log in addition to the line numbers?

Yes, that's on my todo list.

tomverbeure, 2019-01-03:
Another useful feature would be the possibility to only run formal for one particular cover statement, identified by its label. This way, you only get only 1 .vcd file that's always in the same place, irrespective of which cover you're debugging, so I can just keep on using the same gtkwave session.
I currently hack this by putting `if (0) { ... }` around the other cover statements.

cliffordwolf, 2019-03-02:
The label will be the cell name, so you can use script commands to do things such as removing all asserts except the one you want: `delete t:$assert n:foo %d` (see `help select` for details on yosys select expressions)