YosysHQ/sby #28: Yosys+Verific ignores initial RAM/array values

ZipCPU, 2019-03-07:
See included mcve file.

[mcve.zip](https://github.com/YosysHQ/SymbiYosys/files/2942894/mcve.zip)

Within it, two registers, qr[1:0], are set to zero via an initial statement.  They are then asserted to be zero on the initial clock.  The assertion fails when it should not.

Unique to this situation is the fact that qr is defined as a two-item _memory_, even though they are more likely to be implemented as a pair of registers.

Dan

cliffordwolf, 2019-03-08:
> The assertion fails when it should not.

The `logfile.txt` contained in `mcve,zip` says the assertion passes. Did you attach the right file?

ZipCPU, 2019-03-08:
@cliffordwolf ,

Not sure what happened.  I must've packaged up the non-verific version of the bug or some such.  However, when I examined the ZIP file I just sent, it had the "PASS" file within it.  Here's one with a "FAIL" file, and appropriate logfile.

[mcve.zip](https://github.com/YosysHQ/SymbiYosys/files/2945677/mcve.zip)

Dan

cliffordwolf, 2019-03-08:
This looks like a bug in Verific. CC @eddiehung.

Here is a Verific MCVE:

```
module test (
	input clk, i_a, i_d,
	output reg o_d
);
	reg qr [1:0];
	initial	qr[0] = 0;
	initial	qr[1] = 0;

	always @(posedge clk) begin
		o_d <= qr[i_a];
		qr[i_a] <= i_d;
	end
endmodule
```

The value returned by Verific `GetWideInitialValue()` for this memory is `2'b0x` instead of `2'b00`, as directly demonstrated by the output obtained from yosys with the following patch applied.

```
diff --git a/frontends/verific/verific.cc b/frontends/verific/verific.cc
index 9f52ffdc..efe80192 100644
--- a/frontends/verific/verific.cc
+++ b/frontends/verific/verific.cc
@@ -900,6 +900,7 @@ void VerificImporter::import_netlist(RTLIL::Design *design, Netlist *nl, std::se
                        memory->size = number_of_bits / bits_in_word;
 
                        const char *ascii_initdata = net->GetWideInitialValue();
+                       log_dump(ascii_initdata);
                        if (ascii_initdata) {
                                while (*ascii_initdata != 0 && *ascii_initdata != '\'')
                                        ascii_initdata++;
```

The console output for `yosys -p 'verific -sv test.v; verific -import test'`:

```
1. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 07:28:43 2019.
VERIFIC-COMMENT [VERI-1482] Analyzing Verilog file 'test.v'

2. Executing VERIFIC (loading SystemVerilog and VHDL designs using Verific).
Built with Verific Feb19_SW_Release, released at Fri Mar  1 07:28:43 2019.
Adding Verilog module 'test' to elaboration queue.
Running hier_tree::Elaborate().
VERIFIC-INFO [VERI-1018] test.v:1: compiling module 'test'
Importing module test.
DEBUG DUMP IN void Yosys::VerificImporter::import_netlist(RTLIL::Design *, Verific::Netlist *, std::set<Netlist *> &) AT frontends/verific/verific.cc:903:
	ascii_initdata => 2'b0x
```

eddiehung, 2019-03-08:
Filed as VIPER#14502