YosysHQ/sby #37: sby: Failed to create process

trayres, 2019-03-26:
I'm using msys2, and I followed the instructions as best I could - for Z3 I used the released binary package rather than building from source, as the instructions here (https://symbiyosys.readthedocs.io/en/latest/quickstart.html#installing) assume Linux.

The only thing it does is spit out:

> $sby
> failed to create process.

ZipCPU, 2019-03-26:
Try adding a SymbiYosys script to the command line, such as

    sby -f file.sby

One example of such a script would be [this one](https://github.com/ZipCPU/zipcpu/blob/master/bench/formal/ziptimer.sby), used to verify this [fairly simple bus counter](https://github.com/ZipCPU/zipcpu/blob/master/rtl/peripherals/ziptimer.v).  You can find [a longer discussion of this timer, and its proof, here](http://zipcpu.com/zipcpu/2018/04/17/ziptimer.html).

Dan

cliffordwolf, 2019-03-26:
Do you get any output at all from `sby --help`? (Just calling `sby` should read a .sby file from stdin. So maybe that operating mode is not the best thing to start with.)

trayres, 2019-03-26:
sby --help gives me the same error:
failed to create process.

It didn't give me any build errors and is located in 
C:\msys64\usr\local\bin\sby.exe

Hm. Perhaps some combination of things while using MSYS2?

cliffordwolf, 2019-03-27:
Since I'm not a windows user I can only assume this is somehow related to #32. cc @cr1901

trayres, 2019-03-27:
This DOES look related! Excellent that someone else is in the same (or similar) boat.

cr1901, 2019-03-27:
@cliffordwolf Yes, it's related. `failed to create process` is the very helpful and definitely-not-opaque error that [launcher.c](https://github.com/pypa/setuptools/blob/master/launcher.c#L209) emits when it couldn't launch the Python interpreter. `sby.exe` is the compiled version of `launcher.c`.

@trayres Where is the file `sby-script.py` located? Is it in the same directory as `sby.exe`? What does the shebang line of `sby-script.py` look like _after_ installation? `sby.exe` _does_ use the shebang line to find the Python interpreter.

trayres, 2019-03-27:
sby-script.py is located in 

> C:\msys64\usr\local\bin

sby.exe is in the same location

The shebang line of sby-script is the following:
`#!C:/msys64/usr/local/bin/python3`

From a MSYS2 prompt, "where python" produces:
$where python returns:

> C:\msys64\mingw64\bin\python.exe

Replacing the shebang line with #!C:/msys64/mingw64/bin/python.exe returns something now:

> $sby
>   File "C:\msys64\usr\local\bin\sby-script.py", line 130
>     print("ERROR: Use -f to re-run in existing directory.", file=sys.stderr)
>                                                                 ^
> SyntaxError: invalid syntax

cr1901, 2019-03-27:
@trayres Replace with `#!C:/msys64/mingw64/bin/python3.exe`... you appear to be calling the `python2` interpreter.

trayres, 2019-03-27:
Good catch, I thought I had symlinked python to python3 but apparently I had installed them both via MSYS2. 

I will pull latest, reinstall and provide feedback this weekend. Thank you.

cr1901, 2019-03-27:
Like `yosys`, I only intend to support _compiling_ and _installing_ from an MSYS prompt, but users on Windows should be able to _use_ `sby` outside of said prompt. I should probably add something to the `make help` or `README.md` to indicate user-settable `Makefile` vars...

trayres, 2019-03-27:
Just to clarify: Is using from within MSYS2 recommended or viable practice? Or is the intended workflow to compile/install via MSYS2 then use CMD?

cr1901, 2019-03-27:
@trayres If you're using `sby` interactively from a console, then either `msys2` or `cmd` will work fine.

However, stuff like Python's `subprocess` module on Windows (or anything using `CreateProcess` directly) has no concept of `msys2` bash or the shebang; the launcher exectuable called `sby.exe` is meant to workaround this.

trayres, 2019-03-31:
I used this after git pull:
make PREFIX=/mingw64 install

Now I get:
> $sby demo.sby
Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\sby-script.py", line 22, in <module>
    from sby_core import SbyJob, SbyAbort
  File "C:\msys64\mingw64\bin/../share/yosys/python3\sby_core.py", line 37, in <module>
    signal.signal(signal.SIGINT, force_shutdown)
NameError: name 'signal' is not defined

cr1901, 2019-03-31:
@trayres Yes I noticed that too about 2 nights ago; that change appears to postdate my PR being accepted. Anyways I have a new PR in the works to improve Windows support, so I would suggest being on standby until that is ready.

cr1901, 2019-04-03:
@trayres Try [this branch](https://github.com/cr1901/SymbiYosys/tree/async) and see if it fixes your problem :).

trayres, 2019-04-27:
Hm, this is what I get with that branch:

> $ sby demo.sby
> Traceback (most recent call last):
>   File "C:\msys64\mingw64\bin\sby-script.py", line 22, in <module>
>     from sby_core import SbyJob, SbyAbort
>   File [...]

trayres, 2019-04-27:
Then I added "import signal" to sby_core.py, and I got a little farther:

>  $ sby demo.sby
> SBY 16:26:18 [demo] Copy 'demo.sv' to 'demo/src/demo.sv'.
> SBY 16:26:18 [demo] engine_0: smtbmc z3
> SBY 16:26:18 [demo] base: starting process "cd demo/src& yosys -ql ../model/design.log ../model/design.ys"
> Traceback (most recent call last):
>   File "C:\msys64\mingw64\bin\sby-script.py", line 388, in <module>
>     retcode |= run_job(t)
>   File "C:\msys64\mingw64\bin\sby-script.py", line 346, in run_job
>     job.run(setupmode)
>   File "C:\msys64\mingw64\bin/../share/yosys/python3\sby_core.py", line 625, in run
>     self.taskloop()
>   File "C:\msys64\mingw64\bin/../share/yosys/python3\sby_core.py", line 239, in taskloop
>     select(fds, [], [], 1.0) == ([], [], [])
> OSError: [WinError 10093] Either the application has not called WSAStartup, or WSAStartup failed

cr1901, 2019-04-27:
>OSError: [WinError 10093] Either the application has not called WSAStartup, or WSAStartup failed

Mmm, that doesn't seem right at all... line [239](https://github.com/cr1901/SymbiYosys/blob/async/sbysrc/sby_core.py#L239) in my branch isn't a call to select at all... did you forget to do `make PREFIX=/prefix install` before testing?

trayres, 2019-04-27:
I just followed https://symbiyosys.readthedocs.io/en/latest/quickstart.html#installing

cr1901, 2019-04-28:
Line 239 on your machine doesn't match line 239 in the branch that I pointed you to. Can you make sure the `C:\msys64\mingw64\bin\sby-script.py` and `C:\msys64\mingw64\bin/../share/yosys/python3\*.py` files are identical to the [async](https://github.com/cr1901/SymbiYosys/tree/async) branch?

trayres, 2019-04-28:
Something ain't right - it looked like there was another set of files from an earlier attempt to get this up and running. 

So I blew away everything and reinstalled.

> $ sby demo.sby
> Traceback (most recent call last):
>   File "C:\msys64\mingw64\bin\sby-script.py", line 22, in <module>
>     from sby_core import SbyJob, SbyAbort
> ModuleNotFoundError: No module named 'sby_core'

Fine, add it to $PYTHONPATH:
> $ export PYTHONPATH=/c/msys64/usr/local/share/yosys/python3/

Let's try again:
> $ sby demo.sby
> SBY 20:56:17 [demo] Copy 'demo.sv' to 'demo/src/demo.sv'.
> SBY 20:56:18 [demo] engine_0: smtbmc z3
> SBY 20:56:18 [demo] base: starting process "cd demo/src& yosys -ql ../model/design.log ../model/design.ys"
> SBY 20:56:18 [demo] base: finished (returncode=0)
> SBY 20:56:18 [demo] smt2: starting process "cd demo/model& yosys -ql design_smt2.log design_smt2.ys"
> SBY 20:56:18 [demo] smt2: finished (returncode=0)
> SBY 20:56:18 [demo] engine_0: starting process "cd demo& yosys-smtbmc -s z3 --presat --noprogress -t 100 --append 0 --dump-vcd engine_0/trace.vcd --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
> SBY 20:56:18 [demo] engine_0: failed to create process.
> SBY 20:56:18 [demo] engine_0: finished (returncode=0)
> SBY 20:56:18 [demo] ERROR: engine_0: Engine terminated without status.
> SBY 20:56:18 [demo] DONE (ERROR, rc=16)

At this point, I'm sensing that the error is between the screen and the chair, and/or potentially between my ears, however it is now doing more than it previously has. 

Does the (current) error make sense to you, or is it still something wrong with my configuration (...seems likely)?

cr1901, 2019-04-28:
```
SBY 20:56:18 [demo] engine_0: failed to create process.
```

Make sure your `yosys` is up to date- at least newer than [53b28b3](https://github.com/YosysHQ/yosys/commit/53b28b3f01b1ccef0cc5ba0af8000c6013d284bb). This would be one source of that that "very helpful" error message.

trayres, 2019-04-29:
Shoot, I thought it was:
$ yosys -V
Yosys 0.8+369 (git sha1 ea0e0722, x86_64-w64-mingw32-g++ 8.3.0 -Os)

cr1901, 2019-04-29:
@trayres Then make sure `yosys-smtbmc.exe` and `yosys-smtbmc-script.py` exist on your path.

If they exist on your path already, then I don't know what the problem is then, sorry :(.

cmichon, 2024-03-04:
just to share my take on this, as I was facing the same issue today:

```
SBY  0:21:32 [memory] engine_0: starting process "cd memory& yosys-smtbmc -s yices --presat --unroll
 --noprogress -t 10  --append 0 --dump-vcd engine_0/trace.vcd --dump-yw engine_0/trace.yw --dump-vlo
gtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
SBY  0:21:33 [memory] engine_0: failed to create process.
SBY  0:21:33 [memory] engine_0: finished (returncode=0)
SBY  0:21:33 [memory] ERROR: engine_0: Engine terminated without status.
```

I used latest MSYS2 on ucrt64, and discovered that /ucrt64/bin/yosys-smtbmc-script.py and /ucrt64/bin/yosys-witness-script.py contained the following shebang: #!D:/a/msys64/ucrt64/bin/python3.exe

I replaced with the correct path (in my case #!c:/msys64/ucrt64/bin/python3.exe) and now yices engine finally starts properly.

Now I see this:

```
SBY  0:24:03 [memory] engine_0: starting process "cd memory& yosys-smtbmc -s yices --presat --unroll
 --noprogress -t 10  --append 0 --dump-vcd engine_0/trace.vcd --dump-yw engine_0/trace.yw --dump-vlo
gtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
SBY  0:24:04 [memory] engine_0: ##   0:00:00  Solver: yices
SBY  0:24:04 [memory] engine_0: ##   0:00:00  Checking assumptions in step 0..
SBY  0:24:04 [memory] engine_0: ##   0:00:00  Checking assertions in step 0..
SBY  0:24:04 [memory] engine_0: ##   0:00:00  Checking assumptions in step 1..
```