YosysHQ/sby #40: SymbiYosys attempts to use engines not specified in project

litghost, 2019-04-10:
# Expected behavior

SymbiYosys attempts to use engines specified in .sby file

# Actual behavior

If engine specified in .sby fails to find solution, SymbiYosys attempts to use other engines

# Replication instructions:
1. Unzip [symbiyosys.zip]
(https://github.com/YosysHQ/SymbiYosys/files/3066187/symbiyosys.zip)
2. Remove all engines from the path
3. Specifing only yosys and abc, invoke good.sby:
```
sby --yosys $PATH_TO_YOSYS --abc $PATH_TO_ABC good.sby
```

Should complete successfully
4. Specifying only yosys and abc, invoke bad.sby:
```
sby --yosys $PATH_TO_YOSYS --abc $PATH_TO_ABC good.sby
```

sby will first invoke abc as expected, but after failing it will attempt to find smtbmc, etc, etc.

daveshah1, 2019-04-10:
With ABC or AIGER solvers, SymbiYosys needs to invoke an SMT solver to convert the AIGER witness produced for a failure back to a useful counterexample trace. You can disable this with `aigsmt none` in the `[options]` section, see https://symbiyosys.readthedocs.io/en/latest/reference.html

litghost, 2019-04-10:
> With ABC or AIGER solvers, SymbiYosys needs to invoke an SMT solver to convert the AIGER witness produced for a failure back to a useful counterexample trace. You can disable this with `aigsmt none` in the `[options]` section, see https://symbiyosys.readthedocs.io/en/latest/reference.html

What options are accepted for aigsmt?  yices is the default per the doc.

litghost, 2019-04-10:
I've found the section under "currently supported by yosys-smtbmc:"