YosysHQ/sby #43: MSYS2: Makefile with Clang: unsupported option -fPIC for target x86_64-w64-windows-gnu

trayres, 2019-04-19:
Trying to use CONFIG := clang with CXXFLAGS -fPIC on x86_64-w64-windows-gnu gets me:

clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu'

cliffordwolf, 2019-05-01:
Is this a ticket for Yosys? There's no CONFIG nor CXXFLAGS in SymbiYosys so I'm not really sure what to do with this.

(Also, not sure why you are using `CONFIG := clang` and not `CONFIG := msys2` when building with msys. Essentially you are saying you don't enable the msys-specific settings and then wonder why it doesn't work with msys. Afaik there is no `-fPIC` on windows. If you want to use msys2 clang you'd still need to change a few lines in the Makefile, but obviously it's a good idea to start with the configuration for msys and not with the configuration for linux.)

closed 2019-05-01 by cliffordwolf

trayres, 2019-05-01:
+1, should have built with CONFIG := msys2