YosysHQ/sby #44: Warmup failed but there is no assumptions or assertions

minecraft2048, 2019-04-24:
I have a slightly complicated FSM using a 2 process model, and when I try to use sby on it with BMC it failed with warmup failed. 

Here is the smallest test case that I have:
```
module assign2_fsm_behavioral (input clk, input rst, input ped, input car, input T10, input T30, input T60,
                                output reg RM, output reg YM, output reg GM, output reg RS, output reg YS,  output reg GS,
                                output reg trigger);

localparam G_R = 3'd0,
           Y_R = 3'd1,
           R_G = 3'd2,
           R_G_60T = 3'd3,
           R_Y = 3'd4 ;

reg [3:0] state;
reg [3:0] next_state;

//Initial reset

initial begin
    state = 0;
    next_state = 0;
    RM = 0;
    YM = 0;
    GM = 0;
    RS = 0;
    YS = 0;
    GS = 0;
    trigger = 0;
end

always @(*) begin
    case(state)
        G_R: begin
                if ((ped | car) & T60) begin
                    next_state <= Y_R;
                    trigger <= 1;
                    //RM <= 0;
                    YM <= 1;
                    //GM <= 0;
                    //RS <= 1;
                    //YS <= 0;
                    //GS <= 0;                
                end else begin
                    trigger <= 0;
                    next_state <= G_R;
                    //RM <= 0;
                    YM <= 1; // BUG: if this is changed to 0, sby succeeds, but if this is 1, sby fails
                    //GM <= 1;
                    //RS <= 1;    
                    //YS <= 0;
                    //GS <= 0;
                end
        end
        default: begin
                trigger <= 0;
                next_state <= R_G;
                //RM <= 0;
                YM <= 1;
                //GM <= 0;
                //RS <= 0;
                //YS <= 0;
                //GS <= 0;
        end
```

and the sby file:
```
[options]
mode bmc
depth 100

[engines]
smtbmc z3

[script]
read -formal testbug.sv
prep -top assign2_fsm_behavioral

[files]
testbug.sv
 ```

For some reason if

ZipCPU, 2019-04-24:
What's going on is that Yosys treats the logic statements regarding both initial and always @(*) blocks as though they were assumptions.  As a result, you are assuming both that YM = 1 and YM = 0.

Remember, you are verifying "hardware" not "software" or "simulations".  always @(*) YM = 1; is equivalent to stating that YM is tied to power.  initial YM = 0; states that the initial value for a FF should be zero, but YM isn't a FF, hence the problem.

minecraft2048, 2019-04-24:
But if I declare it as a wire then iverilog complains about that YM not a valid l-value and yosys complains 
Warning: wire '\YM' is assigned in a block at testbug.sv:39.

So should I keep it as a reg and remove the initial block, or is there any other way to assign to wire in an always @(*) block?

ZipCPU, 2019-04-25:
So here's the rule:

1. If it's assigned in an always block, regardless of the @() dependencies, it needs to be a reg.
2. If it's assigned on a clock edge, it may have an initial block, otherwise initial blocks don't make any sense.
3. If it's assigned via an "assign" statement, or the output of a module, then it needs to be a "wire"

The warning you saw, "wire '\YM' is assigned in a block ..." is a result of assigning to a wire from within an always @() block.  (Breaking rule 1 above)

Your current code breaks rule 2 above.

Removing the initial statements for anything assigned in an always @(*) block would solve your issue.

Dan

cliffordwolf, 2019-04-29:
> Removing the initial statements for anything assigned in an always @(*) block would solve your issue.

That's the correct answer. :+1:

Closing this issue now.

closed 2019-04-29 by cliffordwolf