YosysHQ/sby #54: Avy solver crashes SES

ZipCPU, 2019-07-23:
Consider the following MCVE:

```verilog
module mcve2(i_clk, o_v);
	input	reg	i_clk;
	output	reg	o_v;

	always @(*)
		o_v = 1;

	always @(*)
		assert(o_v);

endmodule
```

A simple inspection should assure us that this design will pass.

However, when using the SBY file:

```text
[options]
mode prove

[engines]
aiger avy

[script]
read -formal mcve2.v
prep -top mcve2

[files]
mcve2.v
```
When using the 20190416A build of the SymbioticEDA Suite (and I think more recent as well, this is only the one I've got installed), I'm getting the following error when using AVY:

```text

SBY 15:57:52 [mcve2] Removing direcory 'mcve2'.
SBY 15:57:52 [mcve2] Copy 'mcve2.v' to 'mcve2/src/mcve2.v'.
SBY 15:57:52 [mcve2] engine_0: aiger avy
SBY 15:57:52 [mcve2] nomem: starting process "cd mcve2/src; yosys -ql ../model/design_nomem.log ../model/design_nomem.ys"
SBY 15:57:54 [mcve2] nomem: finished (returncode=0)
SBY 15:57:54 [mcve2] aig: starting process "cd mcve2/model; yosys -ql design_aiger.log design_aiger.ys"
SBY 15:57:54 [mcve2] aig: finished (returncode=0)
SBY 15:57:54 [mcve2] engine_0: starting process "cd mcve2; avy --cex - model/design_aiger.aig"
SBY 15:57:57 [mcve2] engine_0: finished (returncode=139)
Traceback (most recent call last):
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/sby", line 388, in <module>
    retcode |= run_job(t)
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/sby", line 346, in run_job
    job.run(setupmode)
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/../share/yosys/python3/sby_core.py", line 624, in run
    self.taskloop()
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/../share/yosys/python3/sby_core.py", line 243, in taskloop
    task.poll()
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/../share/yosys/python3/sby_core.py", line 162, in poll
    self.handle_exit(self.p.returncode)
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/../share/yosys/python3/sby_core.py", line 100, in handle_exit
    self.exit_callback(retcode)
  File "/opt/symbiotic/symbiotic-20190416A/bin/../libexec/../share/yosys/python3/sby_engine_aiger.py", line 82, in exit_callback
    assert task_status is not None
AssertionError
```

## Expected behavior

This design should simply PASS.

Dan

cliffordwolf, 2019-09-18:
Building avy with `cmake -DCMAKE_BUILD_TYPE=Debug` makes avy fail with assertions that reveal that in fact avy is very picky with regard to what AIGER files it accepts. https://github.com/YosysHQ/yosys/commit/779ce3537fa921daf6ffd780264f0867a8ebc4be is a first towards a set of Yosys features that allows us to create "avy compatible" AIGER files.

cliffordwolf, 2019-09-18:
Even with `write_aiger -L` (and without `-O`) I was not able to get avy to process a trivial test case:

```
$ cat test.sv
module test(input a, output o_v);
	assign o_v = a || !a;
	always @(*) assert(o_v);
endmodule

$ yosys -p 'synth; delete -output;; write_aiger -L test.aig' test.sv

$ avy -v9 test.aig
AVY PARAMETERS
	fName = test.aig
	ipt = 0
	avy = 1
	stutter = 0
	reset-cover = 0
	shallow_push = 0
	min-core = 0
	abstraction = 0
	tr0 = 0
	pdr = 100000
	min-suffix = 0
	sat1 = 0
	minisat = 0
	glucose = 1
	minisat_itp = 0
	glucose_itp = 1
	sat_simp = 1
	itp_simp = 0
	proof_reorder = 0
	gen-conf-limit = 0
	kstep = 1
	stick-error = 0
	itp-simplify = 1
	max-frame = 100000
END
Starting ABC
	Reading AIG from 'test.aig'
Found 0 (0) equiv and 1 constans out of 1
Found output driven by intput: 0

************** BRUNCH STATS ***************** 
BRUNCH_STAT run.loop 0.00
BRUNCH_STAT Result UNKNOWN
************** BRUNCH STATS END ***************** 
Assumption size: 1 core size: 1
UNSAT from BMC at frame: 0
itp             : pi =     1  po =     1  and =       0  lev =   0
itp             : pi =     1  po =     1  and =       0  lev =   0

************** BRUNCH STATS ***************** 
BRUNCH_STAT Depth 0
BRUNCH_STAT Frame 1
BRUNCH_STAT OrigItp 0
BRUNCH_STAT SimpItp 0
BRUNCH_STAT Aig_ManSimplifyComb 0.00
BRUNCH_STAT Glucose_solve 0.00
BRUNCH_STAT doBmc 0.00
BRUNCH_STAT getCnfTr 0.00
BRUNCH_STAT getInterpolant 0.00
BRUNCH_STAT run.loop 0.00
BRUNCH_STAT solveWithCore 0.00
BRUNCH_STAT Result UNKNOWN
************** BRUNCH STATS END ***************** 
FINDING NEEDED CONSTRAINTS: CoNum: 1 EquivNum: 1
. Done
Validating ITP: CoNum: 1
.. Done
Building PDR trace
avy: /usr/local/src/extavy/abc/src/misc/vec/vecWec.h:144: abc::Vec_Int_t* abc::Vec_WecEntry(abc::Vec_Wec_t*, int): Assertion `i >= 0 && i < p->nSize' failed.
Aborted (core dumped)
```

I am now considering to simply drop support for AVY in SymbiYosys.