YosysHQ/sby #58: Dockerfile for SymbiYosis

oharboe, 2019-08-08:
I've cobbled together a Dockerfile setup that I intend to use for testing out SymbiYosis based on: https://symbiyosys.readthedocs.io/en/latest/quickstart.html#prerequisites

Would it be an idea to clean it up a bit and add it to SymbiYosis master branch?

[symbiyosis.zip](https://github.com/YosysHQ/SymbiYosys/files/3482018/symbiyosis.zip)

1138-4EB, 2019-08-18:
Refs:

- YosysHQ/yosys#1287
- https://github.com/ghdl/docker/blob/ext/.travis.yml#L29-L41
  - https://hub.docker.com/r/ghdl/synth/tags

oharboe, 2019-08-18:
Looks like this is a solved problem.

Ok to close?

Thanks!

1138-4EB, 2019-08-18:
> Looks like this is a solved problem.
>
> Ok to close?

I wouldn't say so. I just pointed to some alternative references/implementations. Still, there is no Dockerfile in this repo, nor any "official" images at say *hub.docker.com/r/YosysHQ*. Hence, I think this issue should still be open; until we know what the maintainers think about it.

Furthermore, the Dockerfile you proposed is significantly different from the references I provided. Precisely, you used Ubuntu 18.04 as a base, and you included all the solvers (yices2, z3, extavy...), along with some extra tools (e.g. openjdk). You also provided some demo/test scripts. Conversely, the references I provided are meant to be both built and run on CI pipelines (say Travis). Hence, they contain bare minimum resources: either `debian:buster-slim` or `alpine` are used as a base, and z3 is installed only. Since we are targeting different use cases, it'd be sensible to acknowledge this diversity and provide a multi-stage Dockerfile that is configurable enough.