YosysHQ/sby #62: Broken Pipe Error

anmolsahoo25, 2019-09-27:
Can someone please help me out with this? 

System
Ubuntu 18.04
Python3

I am cloning the repository, installing all the pre-reqs and install all the dependencies. Then I,

```bash
cd cores/picorv32
make -C checks
```

The process ends with
```
Traceback (most recent call last):
  File "/usr/local/bin/yosys-smtbmc", line 1418, in <module>
    smt_assert_consequent("(|%s_u| s%d)" % (topmod, step))
  File "/usr/local/bin/yosys-smtbmc", line 1148, in smt_assert_consequent
    smt.write("(assert %s)" % expr)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 402, in write
    stmt = self.unparse(self.unroll_stmt(s))
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 290, in unroll_stmt
    decl = self.unroll_stmt(decl)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 290, in unroll_stmt
    decl = self.unroll_stmt(decl)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 290, in unroll_stmt
    decl = self.unroll_stmt(decl)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 290, in unroll_stmt
    decl = self.unroll_stmt(decl)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in unroll_stmt
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 261, in <listcomp>
    stmt = [self.unroll_stmt(s) for s in stmt]
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 291, in unroll_stmt
    self.write(self.unparse(decl), unroll=False)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 436, in write
    self.p_write(stmt + "\n", True)
  File "/usr/local/bin/../share/yosys/python3/smtio.py", line 319, in p_write
    self.p.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe
```

closed 2019-10-04 by anmolsahoo25

ZipCPU, 2019-10-04:
I see you've closed this issue.  In case anyone else runs into this problem, can you share for us what you learned/discovered?

Dan

anmolsahoo25, 2019-10-04:
Hey! My bad for closing it without a comment. 

I couldn't manage to debug the issue, but it turns out that just installing yosys, symbiyosys and boolector, I was able to get riscv-formal working, so I didn't dig deeper into the issue.

Initially, I was installing the entire stack, yosys, symbiyosys, boolector, z3, yices, avy etc. Maybe I was messing up some step, which caused this issue!

edbordin, 2020-08-02:
If it helps anyone in future - I was hitting this error on OS X and it turned out the root cause was the solver crashing (I was using a pre-built version of Yices 2 which was seemingly not compatible with older versions of OS X)