YosysHQ/sby #66: Create VCD upon pass

noah95, 2019-10-21:
Is it possible to write the vcd even if the design passes? Have not found any help on that. Thanks

ZipCPU, 2019-10-21:
Yes, but there's a special way to do it.

What you want is a `cover()` statement defining the type of vcd trace you are looking for.  (If SymbiYosys did this without the cover statement, you might get a trace of core idling--not very useful)  You'll also need to switch to `mode cover` in your .sby file, or use a task to do both proof and cover in the same sby file.  ([Example here](https://github.com/ZipCPU/zipversa/blob/master/bench/formal/ecpnetdly.sby)).

Typically, I cover the result of every operation my core is supposed to accomplish, and in particular I cover the result of not only the operation completing but the design returning to idle.  (Works great for bus components ...)  I [missed the return to idle check once](http://zipcpu.com/formal/2019/10/05/formal-enough.html), and got myself into a world of hurt, so lesson learned.

Hope this helps, Dan

noah95, 2019-10-22:
Thanks for the reply.

I'm new to formal coming from a functional verification background. To me it is easier to comprehend if I see what the tool is doing using a trace. Currently I try to reproduce your [Simple FiFo Example](https://zipcpu.com/blog/2017/10/19/formal-intro.html).

Changing from `mode bmc` to `mode prove` resulted in the desired vcd dump.

Have to dig deeper into formal to fully understand what's going on under the hood.

Is there any IRC channel to join for these kind of questions? Feels inappropriate to open issues for my questions.

closed 2019-10-22 by noah95

ZipCPU, 2019-10-22:
The simple FIFO example from my first experiences to formal is somewhat outdated.  1) It doesn't use SymbiYosys, 2) it doubles the workload by using and generating its own clock, 3) it never checks the data path, 4) it doesn't generate any cover traces, 5) didn't depend upon proper handshakes to avoid errors, and 6) didn't use the full 2^N elements of the FIFO but rather just (2^N)-1.

Check out the FIFO lesson in the [tutorial](https://zipcpu.com/tutorial).  It's a better example, fixing the bugs listed above, and even has some decent cover traces going along with it.

Dan

ZipCPU, 2019-10-22:
In hindsight it looks like I didn't answer your IRC question.  Yes, there is an IRC channel you can go to.  The freenode #yosys channel is usually pretty good.

We can chat more there if you would like.

Dan