YosysHQ/sby #75: Work directory could contain 'sby' in name to allow for a single '.gitignore' rule

tudortimi, 2020-04-04:
When starting 'sby -f some_setup.sby', sby will create a work directory called 'some_setup'. This directory contains scratch data and should not be checked into source control. For example, for Git, a line with 'some_setup/' would be needed in '.gitignore'. One such line would be needed for each 'sby' file. Renaming the 'sby' file would mean that the name of the directory changes and would also require a change in '.gitignore'. It would be much easier to ignore all of these work directories if they matched some similar expression. For example, they could all be called 'sby.<name of sby file>'. This would mean that 'sby.*/' in '.gitignore' would be sufficient.

tudortimi, 2020-04-04:
Alternatively, we could also store work directories under an 'sby_work' directory and ignore that. Other tools also provide a command line switch to select the name of the work directory.

tmeissner, 2020-04-04:
I use the following command in my Makefiles to get a work directory work/ with seperate directories for my cover/bmc/prove runs in it:

```
cover bmc prove: ${DUT}.vhd symbiyosys.sby
	sby --yosys "yosys -m ghdl" -f -d work/${DUT}-$@ symbiyosys.sby $@
```

ZipCPU, 2020-04-10:
@tudortimi : It looks like the `-d` option supports what you are looking for.

@tmeissner : I didn't realize `sby` supported a `-d` option.  Thanks for pointing it out to me!

Dan

closed 2020-04-10 by cliffordwolf

tudortimi, 2020-04-10:
I was thinking more along the lines of not needing the '-d' option to get a "stable" output directory.

ZipCPU, 2020-04-10:
Yeah, I get it.  I'm also thinking about all of the files I'd need to change now that I've come to depend upon SymbiYosys' current behavior.

Are you running SymbiYosys from a make file?  If you did so, it would 1) make it easier to add a `-d <dir>` option to your `sby` calls, and 2) make it easier to handle multiple proofs in separate processes from a single command line invocation.

tudortimi, 2020-04-10:
I'm running using sby directly. I would have wanted to have a generic .gitignore that would work in the case I don't mess around with '-d'. Something like the one there is for Java: https://github.com/github/gitignore/blob/master/Java.gitignore Using '-d' would have, of course, forced me to add a line to .gitignore myself.