YosysHQ/sby #82: Can't set value of enum parameter using 'chparam'

tudortimi, 2020-04-12:
I have the `example.sv` file:

```
module example;

  parameter enum { FOO, BAR } some_param = FOO;

endmodule
```

and `example.sby`:

```
[options]
mode cover

[engines]
smtbmc

[script]
read -verific
read -formal example.sv
hierarchy -chparam some_param BAR -top example
prep

[files]
example.sv
```
I would like to set the value of `some_param` to `BAR` via the `script` block.

I get the following error: `ERROR: Can't decode value 'BAR'!`.

I've also tried quoting the parameter value as `"BAR"`, but then I get `ERROR: Can't decode value '"BAR"'!` (notice the extra quotes around `BAR`).

With single quotes (`'BAR'`) I get `input:0: ERROR: Digit larger than 1 used in in base-2 constant.`.

jimmysitu, 2021-05-01:
Hi, All

I got similar issue
```verilog
module priority_encoder #
(
    // LSB priority: "LOW", "HIGH"
    parameter LSB_PRIORITY = "LOW"
)
```
```sby
[script]
read -formal priority_encoder.v
if ("high" in tags):
    cmd += " -chparam LSB_PRIORITY HIGH"
output(cmd)
prep -top priority_encoder
[files]
priority_encoder.v
```
I got the error
```bash
ERROR: Can't decode value 'HIGH'!
```

jimmysitu, 2021-05-01:
Hi, @tudortimi 

I fixed by problem with `chparam -set` in yosys. May be you can try
```
[script]
read -verific
read -formal example.sv
chparam -set some_param "BAR"
hierarchy -top example
prep
```