YosysHQ/sby #91: vcd file includes blank traces

mattvenn, 2020-06-12:
I recall previous behaviour was that untouched signals were dropped from the vcd. Now untouched signals are present in the vcd but have no value - which IMO is more confusing.

```
`default_nettype        none
module counter (input wire clock);

    reg [2:0] f_counter1 = 0;
    reg [2:0] f_counter2 = 0;

        always @(posedge clock) begin
        f_counter1 <= f_counter1 + 1;
        f_counter2 <= f_counter2 + 1;
        assert(f_counter2 != 3);
    end

endmodule
```

```
[options]
mode prove
depth 5

[engines]
smtbmc

[script]
read -formal counter.v
prep -top counter

[files]
counter.v
```

[image]

[counter.zip](https://github.com/YosysHQ/SymbiYosys/files/4770353/counter.zip)

ZipCPU, 2020-06-12:
Which signal in the trace above are you referencing?  f_counter1?  or some other signal?

mattvenn, 2020-06-12:
Yes

ZipCPU, 2020-06-12:
I'm confused then.  `f_counter1` shows in your trace.  It shows as the low line rather than both simply because this is considered an "upgrade" of gtkwave.  You can see this if you examine `f_counter2` and see what happens when it switches to all zeros.

mattvenn, 2020-06-15:
f_counter1 shows in the trace but there is no data. It is incrementing every time like f_counter2. 

Previously I've noticed that signals that are not used in the formal process are not present in the vcd (unless using *keep*).

Here is an example of an unused signal trace being kept but containing no data.

ZipCPU, 2020-06-15:
Ok, I see it now, it's not that `f_counter1` isn't present, it's that it has the wrong value.

Yeah, that looks like a bug.

Dan

nakengelhardt, 2020-06-16:
It's the call to `opt_clean` in preparing the model that does this; there were a bunch of changes to `opt_clean` recently but before I track down which one, I'd rather understand first if it's doing what it is supposed to, and what the preferred outcome would be.

If you just want `f_counter1` to disappear again, changing it to `opt_clean -purge` would do that. But I think the better solution would be to treat any public net like an output when creating the model, so that both it and its behavior is kept. @clairexen is there an easy way to do that?

nakengelhardt, 2020-06-23:
Here's an ugly hotfix for if you want proper traces:
```
[script]
read -formal counter.v
hierarchy -top counter
proc
setattr -set keep 1 w:\\*
prep
```