YosysHQ/sby #105: BMC violates assumptions with multiclock

jleahy, 2020-07-30:
I have a simple example below:

```
module test;

(* gclk *) reg gclk;

(* anyseq *) reg x;
(* anyseq *) reg y;

initial assume(!x);
initial assume(!y);

always @(posedge gclk) begin
    assume(!$fell(x));
    assume(!$fell(y));
    assume($rose(x) == $rose(y));
end

always @(*) begin
    assert(x == y);
end

endmodule
```

And I run this using the following sby:

```
[options]
mode bmc
depth 10
multiclock on

[engines]
smtbmc boolector

[script]
read_verilog -formal test.v
prep -top test

[files]
test.v
```

I would expect this to succeed. Both variables start off low, and they are not allowed to fall (only rise). We assume that they rise at the same time and thus we can assert that they are always equal. However the produced trade shows `x` rising while `y` stays low on step 1.

It is possible that I'm doing something horribly wrong, of course.

This is running from master, I can provide the smt2 if it turns out to be useful.

ZipCPU, 2020-07-30:
This behavior is by design.

The issue is that the `always @(*)` assertion is evaluated before the `always @(posedge clk)` assumptions are applied.  If you change your `always @(*)` assertion to be an `always @(posedge clk)` assertion, the design will pass.

To understand what's going on, you need a bit of an understanding of an `always @(posedge clk)` property when compared to an `always @(*)` property.  The issue is that the `always @(posedge clk)` properties aren't applied until _after_ the clock cycle has completed, whereas `always @(*)` properties apply immediately--even before any related `always @(posedge clk)` properties will be applied.

Dan

jleahy, 2020-07-30:
I see, that's unfortunate as I don't think the `always @(posedge clk)` assertions can be used for induction, only `always @(*)`. However if that matches the spec for systemverilog then there's not much that can be done about it.

Essentially, if I understand you correctly, there's no way to restrict traces to ones where x and y rise in the same time step (this is in the context of multiple asynchronous resets that are driven from a single source). Instead what I need to do is ensure that my asserts can handle the cases where one reset has risen but the other has not yet, and instead add asserts to make sure that a reset cannot fall until the partner reset has also risen.

ZipCPU, 2020-07-30:
`always @(posedge clk)` assertions can most definitely be used with induction.  I do so all the time.

> ... there's no way to restrict traces to ones where x and y rise in the same time step ...

Yes there is.  In general, you can either 1) implement everything on the global clock, with no `@(*)` assertions, or 2) implement your own versions of the SystemVerilog `$past()` family of functions so they don't have this problem.  Feel free to check out [my article on verifying an asynchronous FIFO](https://zipcpu.com/blog/2018/07/06/afifo.html).  I deal with the topic there.  I've since [rebuilt the FIFO from the ground up](https://github.com/ZipCPU/wb2axip/blob/master/rtl/afifo.v), and the updated proof only takes about 5-seconds (vs nearly an hour).  The difference between the two IIRC is the difference between `@(posedge clk)` and `@(*)`.

Dan

jleahy, 2020-07-30:
Ok, apologies for an unnecessary bug report. I find moving lots asserts into clocked blocks somewhat distasteful, so I'll instead take the 'defining a custom past' route which seems to work fine. It looks like that's what you've done in your rebuilt FIFO also.

For the benefits of any people passing by from Google I'll post the fixed version below.

```
module test;

(* anyseq *) reg x;
(* anyseq *) reg y;

(* gclk *) reg gclk;
reg is_init = 1'b1;
reg prev_x;
reg prev_y;
always @(posedge gclk) begin
    is_init <= 1'b0;
    prev_x <= x;
    prev_y <= y;
end

wire fell_x = !x && prev_x;
wire fell_y = !y && prev_y;
wire rose_x = x && !prev_x;
wire rose_y = y && !prev_y;
always @(*) begin
    if (is_init) begin
        assume(!x);
        assume(!y);
    end else begin
        assume(!fell_x);
        assume(!fell_y);
        assume(rose_x == rose_y);
    end
end

always @(*) begin
    assert(x == y);
end

endmodule
```

Thanks for the help!

closed 2020-07-30 by jleahy