YosysHQ/sby #110: No such command: dffunmap

thirtythreeforty, 2020-08-14:
I have installed SymbiYosys from the AUR -git package.

I receive the following error when trying to BMC a simple design:

```
[georgev@desertvoice rtltools (master)]$ sby pipeshift.sby 
SBY 17:52:50 [pipeshift] Copy 'pipeshift.v' to 'pipeshift/src/pipeshift.v'.
SBY 17:52:50 [pipeshift] engine_0: smtbmc
SBY 17:52:50 [pipeshift] base: starting process "cd pipeshift/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 17:52:50 [pipeshift] base: finished (returncode=0)
SBY 17:52:50 [pipeshift] smt2: starting process "cd pipeshift/model; yosys -ql design_smt2.log design_smt2.ys"
SBY 17:52:50 [pipeshift] smt2: ERROR: No such command: dffunmap (type 'help' for a command overview)
SBY 17:52:50 [pipeshift] smt2: finished (returncode=1)
SBY 17:52:50 [pipeshift] smt2: job failed. ERROR.
SBY 17:52:50 [pipeshift] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:52:50 [pipeshift] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:52:50 [pipeshift] DONE (ERROR, rc=16)
```

It isn't clear why sy is running this command, or where I get it from - the recent commit is the only mention on Google. How do I install dffunmap?

thirtythreeforty, 2020-08-14:
Whoops, dupe of #107, sorry for the noise.

closed 2020-08-14 by thirtythreeforty

nickoe, 2021-01-24:
@thirtythreeforty Hmm, I am having the exact same issue. Issue #107 mentions that we need to use master, but the AUR package is indeed doing this. Were you able to resolve this? I am on 091222b87febb10fad87fcbe98a57599a54c5fd3.

nickoe, 2021-01-24:
Oohhh, yosys master and not SymbiYosys master. :D