YosysHQ/sby #122: tests/both_ex.sby verification fails with 'btorsim' error: invalid command line option '--vcd'

satnam6502, 2021-05-03:
Repo, using the Docker image I created for SymbiYosys:

```console
$ cat ~/symbiyosys/sby.sh
#!/bin/bash
docker run --rm -v $(pwd):$(pwd) -u $(id -u) --workdir=$(pwd) satnam6502/symbiyosys $@
$ ~/symbiyosys/sby.sh both_ex.sby
SBY 18:15:55 [both_ex_btormc] Copy 'both_ex.v' to 'both_ex_btormc/src/both_ex.v'.
SBY 18:15:55 [both_ex_btormc] engine_0: btor btormc
SBY 18:15:55 [both_ex_btormc] base: starting process "cd both_ex_btormc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 18:15:55 [both_ex_btormc] base: finished (returncode=0)
SBY 18:15:55 [both_ex_btormc] btor: starting process "cd both_ex_btormc/model; yosys -ql design_btor.log design_btor.ys"
SBY 18:15:56 [both_ex_btormc] btor: finished (returncode=0)
SBY 18:15:56 [both_ex_btormc] engine_0: starting process "cd both_ex_btormc; btormc --stop-first 1 -v 1 -kmax 4 model/design_btor.btor"
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] calling BMC on 1 properties from bound 0 up-to maximum bound k = 4
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 0 UNSATISFIABLE
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 0
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 1 UNSATISFIABLE
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 1
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 2 UNSATISFIABLE
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 2
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 3 UNSATISFIABLE
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 3
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 4 UNSATISFIABLE
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 4
SBY 18:15:56 [both_ex_btormc] engine_0: [btor>mc] deleting model checker: 2 inputs, 4 states, 1 bad, 1 constraints
SBY 18:15:56 [both_ex_btormc] engine_0: finished (returncode=0)
SBY 18:15:56 [both_ex_btormc] engine_0: Status returned by engine: pass
SBY 18:15:56 [both_ex_btormc] engine_0: Engine did not produce a counterexample.
SBY 18:15:56 [both_ex_btormc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:15:56 [both_ex_btormc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:15:56 [both_ex_btormc] summary: engine_0 (btor btormc) returned pass
SBY 18:15:56 [both_ex_btormc] DONE (PASS, rc=0)
SBY 18:15:56 [both_ex_pono] Copy 'both_ex.v' to 'both_ex_pono/src/both_ex.v'.
SBY 18:15:56 [both_ex_pono] engine_0: btor pono
SBY 18:15:56 [both_ex_pono] base: starting process "cd both_ex_pono/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 18:15:56 [both_ex_pono] base: finished (returncode=0)
SBY 18:15:56 [both_ex_pono] btor: starting process "cd both_ex_pono/model; yosys -ql design_btor.log design_btor.ys"
SBY 18:15:56 [both_ex_pono] btor: finished (returncode=0)
SBY 18:15:56 [both_ex_pono] engine_0: starting process "cd both_ex_pono; pono -v 1 -e bmc -k 4 model/design_btor.btor"
SBY 18:15:56 [both_ex_pono] engine_0: bash: pono: command not found
SBY 18:15:56 [both_ex_pono] engine_0: finished (returncode=127)
SBY 18:15:56 [both_ex_pono] engine_0: COMMAND NOT FOUND. ERROR.
SBY 18:15:56 [both_ex_pono] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:15:56 [both_ex_pono] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:15:56 [both_ex_pono] DONE (ERROR, rc=16)
SBY 18:15:56 [both_ex_cover] Copy 'both_ex.v' to 'both_ex_cover/src/both_ex.v'.
SBY 18:15:56 [both_ex_cover] engine_0: btor btormc
SBY 18:15:56 [both_ex_cover] base: starting process "cd both_ex_cover/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 18:15:56 [both_ex_cover] base: finished (returncode=0)
SBY 18:15:56 [both_ex_cover] btor: starting process "cd both_ex_cover/model; yosys -ql design_btor.log design_btor.ys"
SBY 18:15:56 [both_ex_cover] btor: finished (returncode=0)
SBY 18:15:56 [both_ex_cover] engine_0: starting process "cd both_ex_cover; btormc --stop-first 0 -v 1 -kmax 4 model/design_btor.btor"
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] calling BMC on 2 properties from bound 0 up-to maximum bound k = 4
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 0 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 0 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 0
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 1 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 1 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 1
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 2 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 2 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 2
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 0 reachable at bound k = 3 SATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0_0: starting process "cd both_ex_cover ; btorsim -c --vcd engine_0/trace0.vcd --hierarchical-symbols --info model/design_btor.info model/design_btor.btor engine_0/trace0.wit"
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 3 UNSATISFIABLE
SBY 18:15:56 [both_ex_cover] engine_0: [btor>mc] found 1 reachable and 0 unreachable bad state properties at bound k = 3
SBY 18:15:56 [both_ex_cover] engine_0_0: *** 'btorsim' error: invalid command line option '--vcd' (try '-h')
SBY 18:15:56 [both_ex_cover] engine_0_0: finished (returncode=1)
Traceback (most recent call last):
  File "/usr/local/bin/sby", line 430, in <module>
    retcode |= run_job(t)
  File "/usr/local/bin/sby", line 388, in run_job
    job.run(setupmode)
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 665, in run
    self.taskloop()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 278, in taskloop
    task.poll()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 191, in poll
    self.handle_exit(self.p.returncode)
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 115, in handle_exit
    self.exit_callback(retcode)
  File "/usr/local/bin/../share/yosys/python3/sby_engine_btor.py", line 113, in exit_callback2
    assert retcode == 0
AssertionError
```

satnam6502, 2021-05-03:
So after installing `btorsim` I now get as far as:
```console
SBY 18:49:33 [both_ex_cover] engine_0: [btor>mc] bad state property 0 reachable at bound k = 3 SATISFIABLE
SBY 18:49:33 [both_ex_cover] engine_0_0: starting process "cd both_ex_cover ; btorsim -c --vcd engine_0/trace0.vcd --hierarchical-symbols --info model/design_btor.info model/design_btor.btor engine_0/trace0.wit"
SBY 18:49:33 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 3 UNSATISFIABLE
SBY 18:49:33 [both_ex_cover] engine_0: [btor>mc] found 1 reachable and 0 unreachable bad state properties at bound k = 3
SBY 18:49:33 [both_ex_cover] engine_0_0: btorsim: error while loading shared libraries: libbtor2parser.so: cannot open shared object file: No such file or directory
SBY 18:49:33 [both_ex_cover] engine_0_0: finished (returncode=127)
SBY 18:49:33 [both_ex_cover] engine_0_0: COMMAND NOT FOUND. ERROR.
SBY 18:49:33 [both_ex_cover] engine_0: terminating process
SBY 18:49:33 [both_ex_cover] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:49:33 [both_ex_cover] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:49:33 [both_ex_cover] DONE (ERROR, rc=16)
SBY 18:49:33 One or more tasks produced a non-zero return code.
```
Let's see what I can do about this...

satnam6502, 2021-05-03:
After building btor2tools off GitHib with `--static` I got this to go through.
```console
satnam@satnam-glaptop:~/sy/tests$ sudo ~/symbiyosys/sby.sh -f both_ex.sby
[sudo] password for satnam: 
SBY 20:59:46 [both_ex_btormc] Removing directory 'both_ex_btormc'.
SBY 20:59:46 [both_ex_btormc] Copy 'both_ex.v' to 'both_ex_btormc/src/both_ex.v'.
SBY 20:59:46 [both_ex_btormc] engine_0: btor btormc
SBY 20:59:46 [both_ex_btormc] base: starting process "cd both_ex_btormc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 20:59:46 [both_ex_btormc] base: finished (returncode=0)
SBY 20:59:46 [both_ex_btormc] btor: starting process "cd both_ex_btormc/model; yosys -ql design_btor.log design_btor.ys"
SBY 20:59:46 [both_ex_btormc] btor: finished (returncode=0)
SBY 20:59:46 [both_ex_btormc] engine_0: starting process "cd both_ex_btormc; btormc --stop-first 1 -v 1 -kmax 4 model/design_btor.btor"
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] calling BMC on 1 properties from bound 0 up-to maximum bound k = 4
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 0 UNSATISFIABLE
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 0
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 1 UNSATISFIABLE
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 1
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 2 UNSATISFIABLE
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 2
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 3 UNSATISFIABLE
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 3
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 4 UNSATISFIABLE
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 4
SBY 20:59:46 [both_ex_btormc] engine_0: [btor>mc] deleting model checker: 2 inputs, 4 states, 1 bad, 1 constraints
SBY 20:59:46 [both_ex_btormc] engine_0: finished (returncode=0)
SBY 20:59:46 [both_ex_btormc] engine_0: Status returned by engine: pass
SBY 20:59:46 [both_ex_btormc] engine_0: Engine did not produce a counterexample.
SBY 20:59:46 [both_ex_btormc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_btormc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_btormc] summary: engine_0 (btor btormc) returned pass
SBY 20:59:46 [both_ex_btormc] DONE (PASS, rc=0)
SBY 20:59:46 [both_ex_pono] Removing directory 'both_ex_pono'.
SBY 20:59:46 [both_ex_pono] Copy 'both_ex.v' to 'both_ex_pono/src/both_ex.v'.
SBY 20:59:46 [both_ex_pono] engine_0: btor pono
SBY 20:59:46 [both_ex_pono] base: starting process "cd both_ex_pono/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 20:59:46 [both_ex_pono] base: finished (returncode=0)
SBY 20:59:46 [both_ex_pono] btor: starting process "cd both_ex_pono/model; yosys -ql design_btor.log design_btor.ys"
SBY 20:59:46 [both_ex_pono] btor: finished (returncode=0)
SBY 20:59:46 [both_ex_pono] engine_0: starting process "cd both_ex_pono; pono -v 1 -e bmc -k 4 model/design_btor.btor"
SBY 20:59:46 [both_ex_pono] engine_0: bash: pono: command not found
SBY 20:59:46 [both_ex_pono] engine_0: finished (returncode=127)
SBY 20:59:46 [both_ex_pono] engine_0: COMMAND NOT FOUND. ERROR.
SBY 20:59:46 [both_ex_pono] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_pono] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_pono] DONE (ERROR, rc=16)
SBY 20:59:46 [both_ex_cover] Removing directory 'both_ex_cover'.
SBY 20:59:46 [both_ex_cover] Copy 'both_ex.v' to 'both_ex_cover/src/both_ex.v'.
SBY 20:59:46 [both_ex_cover] engine_0: btor btormc
SBY 20:59:46 [both_ex_cover] base: starting process "cd both_ex_cover/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 20:59:46 [both_ex_cover] base: finished (returncode=0)
SBY 20:59:46 [both_ex_cover] btor: starting process "cd both_ex_cover/model; yosys -ql design_btor.log design_btor.ys"
SBY 20:59:46 [both_ex_cover] btor: finished (returncode=0)
SBY 20:59:46 [both_ex_cover] engine_0: starting process "cd both_ex_cover; btormc --stop-first 0 -v 1 -kmax 4 model/design_btor.btor"
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] calling BMC on 2 properties from bound 0 up-to maximum bound k = 4
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 0 UNSATISFIABLE
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 0 UNSATISFIABLE
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 0
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 1 UNSATISFIABLE
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 1 UNSATISFIABLE
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 1
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 0 at bound k = 2 UNSATISFIABLE
SBY 20:59:46 [both_ex_cover] engine_0: [btor>mc] bad state property 1 at bound k = 2 UNSATISFIABLE
[... cut ...]
SBY 20:59:46 [both_ex_cover] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_cover] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:59:46 [both_ex_cover] summary: engine_0 (btor btormc) returned pass
SBY 20:59:46 [both_ex_cover] summary: trace: both_ex_cover/engine_0/trace0.vcd
SBY 20:59:46 [both_ex_cover] summary: trace: both_ex_cover/engine_0/trace1.vcd
SBY 20:59:46 [both_ex_cover] DONE (PASS, rc=0)
SBY 20:59:46 One or more tasks produced a non-zero return code.
```

satnam6502, 2021-05-03:
But `mixed` does not work due to an error to do with `pono`:

```console
$ ~/symbiyosys/sby.sh -f mixed.sby
SBY 21:07:37 [mixed_cover] Removing directory 'mixed_cover'.
SBY 21:07:37 [mixed_cover] Copy 'mixed.v' to 'mixed_cover/src/mixed.v'.
SBY 21:07:37 [mixed_cover] engine_0: btor btormc
SBY 21:07:37 [mixed_cover] base: starting process "cd mixed_cover/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 21:07:37 [mixed_cover] base: finished (returncode=0)
SBY 21:07:37 [mixed_cover] btor: starting process "cd mixed_cover/model; yosys -ql design_btor.log design_btor.ys"
SBY 21:07:37 [mixed_cover] btor: finished (returncode=0)
SBY 21:07:37 [mixed_cover] engine_0: starting process "cd mixed_cover; btormc --stop-first 0 -v 1 -kmax 19 model/design_btor.btor"
SBY 21:07:37 [mixed_cover] engine_0: [btor>mc] calling BMC on 2 properties from bound 0 up-to maximum bound k = 19
SBY 21:07:37 [mixed_cover] engine_0: [btor>mc] bad state property 0 reachable at bound k = 0 SATISFIABLE
SBY 21:07:37 [mixed_cover] engine_0_0: starting process "cd mixed_cover ; btorsim -c --vcd engine_0/trace0.vcd --hierarchical-symbols --info model/design_btor.info model/design_btor.btor engine_0/trace0.wit"
SBY 21:07:37 [mixed_cover] engine_0: [btor>mc] bad state property 1 reachable at bound k = 0 SATISFIABLE
SBY 21:07:37 [mixed_cover] engine_0_1: starting process "cd mixed_cover ; btorsim -c --vcd engine_0/trace1.vcd --hierarchical-symbols --info model/design_btor.info model/design_btor.btor engine_0/trace1.wit"
SBY 21:07:37 [mixed_cover] engine_0: [btor>mc] found 2 reachable and 0 unreachable bad state properties at bound k = 0
SBY 21:07:37 [mixed_cover] engine_0: [btor>mc] deleting model checker: 5 inputs, 4 states, 2 bad, 2 constraints
SBY 21:07:37 [mixed_cover] engine_0: finished (returncode=0)
SBY 21:07:37 [mixed_cover] engine_0_0: finished (returncode=0)
SBY 21:07:37 [mixed_cover] engine_0_1: finished (returncode=0)
SBY 21:07:37 [mixed_cover] engine_0: Status returned by engine: pass
SBY 21:07:37 [mixed_cover] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_cover] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_cover] summary: engine_0 (btor btormc) returned pass
SBY 21:07:37 [mixed_cover] summary: trace: mixed_cover/engine_0/trace0.vcd
SBY 21:07:37 [mixed_cover] summary: trace: mixed_cover/engine_0/trace1.vcd
SBY 21:07:37 [mixed_cover] DONE (PASS, rc=0)
SBY 21:07:37 [mixed_btormc] Removing directory 'mixed_btormc'.
SBY 21:07:37 [mixed_btormc] Copy 'mixed.v' to 'mixed_btormc/src/mixed.v'.
SBY 21:07:37 [mixed_btormc] engine_0: btor btormc
SBY 21:07:37 [mixed_btormc] base: starting process "cd mixed_btormc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 21:07:37 [mixed_btormc] base: finished (returncode=0)
SBY 21:07:37 [mixed_btormc] btor: starting process "cd mixed_btormc/model; yosys -ql design_btor.log design_btor.ys"
SBY 21:07:37 [mixed_btormc] btor: finished (returncode=0)
SBY 21:07:37 [mixed_btormc] engine_0: starting process "cd mixed_btormc; btormc --stop-first 1 -v 1 -kmax 0 model/design_btor.btor"
SBY 21:07:37 [mixed_btormc] engine_0: [btor>mc] calling BMC on 1 properties from bound 0 up-to maximum bound k = 0
SBY 21:07:37 [mixed_btormc] engine_0: [btor>mc] bad state property 0 at bound k = 0 UNSATISFIABLE
SBY 21:07:37 [mixed_btormc] engine_0: [btor>mc] found 0 reachable and 0 unreachable bad state properties at bound k = 0
SBY 21:07:37 [mixed_btormc] engine_0: [btor>mc] deleting model checker: 5 inputs, 4 states, 1 bad, 2 constraints
SBY 21:07:37 [mixed_btormc] engine_0: finished (returncode=0)
SBY 21:07:37 [mixed_btormc] engine_0: Status returned by engine: pass
SBY 21:07:37 [mixed_btormc] engine_0: Engine did not produce a counterexample.
SBY 21:07:37 [mixed_btormc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_btormc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_btormc] summary: engine_0 (btor btormc) returned pass
SBY 21:07:37 [mixed_btormc] DONE (PASS, rc=0)
SBY 21:07:37 [mixed_pono] Removing directory 'mixed_pono'.
SBY 21:07:37 [mixed_pono] Copy 'mixed.v' to 'mixed_pono/src/mixed.v'.
SBY 21:07:37 [mixed_pono] engine_0: btor pono
SBY 21:07:37 [mixed_pono] base: starting process "cd mixed_pono/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 21:07:37 [mixed_pono] base: finished (returncode=0)
SBY 21:07:37 [mixed_pono] btor: starting process "cd mixed_pono/model; yosys -ql design_btor.log design_btor.ys"
SBY 21:07:37 [mixed_pono] btor: finished (returncode=0)
SBY 21:07:37 [mixed_pono] engine_0: starting process "cd mixed_pono; pono -v 1 -e bmc -k 0 model/design_btor.btor"
SBY 21:07:37 [mixed_pono] engine_0: bash: pono: command not found
SBY 21:07:37 [mixed_pono] engine_0: finished (returncode=127)
SBY 21:07:37 [mixed_pono] engine_0: COMMAND NOT FOUND. ERROR.
SBY 21:07:37 [mixed_pono] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_pono] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 21:07:37 [mixed_pono] DONE (ERROR, rc=16)
SBY 21:07:37 One or more tasks produced a non-zero return code.
```

satnam6502, 2021-05-03:
I tried to install Pono but failed so I will just avoid using the pono engine :-(
```
# Pono
RUN git clone https://github.com/upscale-project/pono.git && \
    cd pono && ./contrib/setup-smt-switch.sh && \
    ./configure.sh && cd build && make
```
did not work.

mmicko, 2021-05-04:
@satnam6502 Sorry I have not  answer before, but I think we have something that can help you and save you some time. 

```
git clone -b packages https://github.com/YosysHQ/fpga-nightly
cd fpga-nightly
./nightly.py build  --target=minimal --arch=linux-x64
```

In `_outputs/linux-x64/minimal` you will have files ready to be used on any x64 based Linux (all dependencies are in) so that can be just added to your docker image as is, and just add location to PATH.
This minimal target includes just yosys, sby and solvers, please note cvc4 takes time to build.
These scripts will detect if anything changes on used repositories so will rebuild only if there is need, so it is good if you need to maintain it for your local usage as well. Please note that branch used is still in works, and I will do some changes during time, but it is more usable for your case then master currently is.
In case compiling all required packages is too time consuming for you or you have any other issue, please contact me directly (email visible in github account page).

nakengelhardt, 2022-01-11:
This was due to btorsim --vcd being a feature I added, which was not at the time upstreamed yet. Closing as it was merged upstream a while ago.

closed 2022-01-11 by nakengelhardt