YosysHQ/sby #137: pono only checks first property in a design

nakengelhardt, 2022-01-12:
When using the pono solver, it only checks one property. It will mention this at the beginning, e.g. `Solving property: both_ex.v:19.25-20.30`.

We don't have a testcase that exposes this problem yet.

nakengelhardt, 2022-01-12:
Testcase:

```
[tasks]
btormc
pono

[options]
mode bmc
depth 5
expect fail

[engines]
btormc: btor btormc
pono: btor pono

[script]
read_verilog -sv multi_assert.v
prep -top test

[file multi_assert.v]
module test();
always @* begin
assert (1);
assert (0);
end
endmodule
```

tmeissner, 2022-01-12:
I also have stumbled over this behavior some time ago, but was not sure, if that is by intention. `pono` has an option to set the index of the property to check, but I didn't found any option to solve all properties.

nakengelhardt, 2022-01-12:
Yes, looking at the options code for pono the intention seems to be not to check more than one property at a time. Yosys' `write_btor` does have an option for creating a single `bad` that is an and of all the assertions. It'll do as a workaround until I can figure out a better solution, but it removes the ability to find out which property failed so that's not great either.

PR #136 (2022-01-12): use --witness option when calling pono
The pono solver interface has changed a bit since we added support for it. [Printing the witness is now no longer automatic](https://github.com/upscale-project/pono/pull/210), it requires passing the `--witness` option, so we should do that.

Pono also only checks the first property in a design (#137); this adds a testcase exposing the issue and provides a workaround that sort-of fixes #137 but at the cost of no longer being able to tell which property failed.
closed 2022-01-12 by mmicko