YosysHQ/sby #157: SymbiYosys tags and known working version of Yosys

vmunoz82, 2022-04-28:
Hi!
Trying the latest (0.16) tag of yosys with the master sby version #156 I got the following error:

`"base: ERROR: No such command: write_jny (type 'help' for a command overview)"`

As such yosys version don't have yet write_jny command, I found that going back to 4886ed7e1999fd24a6e4d4383bc4fa1f91c2484e works OK for me.

The latest commit on master of sby is supposed to work on pair with the latest commit of sby master version?

nakengelhardt, 2022-05-20:
Yes, it is intended to work if you take versions of both from the same day (as the nightly build does). When changes are required on both sides, we usually merge modifications in yosys first and sby the next day (when the sby CI has received the new nightly yosys build from OSS CAD Suite and the tests pass).

lethalbit, 2022-08-10:
Closing as the latest tagged version of Yosys supports the `write_jny` command now.

closed 2022-08-10 by lethalbit