YosysHQ/sby #208: [RFC] Ensuring code is formatted in accordance to pep8

lethalbit, 2022-08-04:
I suggest that we add as a development check to ensure the code is formatted in accordance with pep8 to ensure consistency.

I use [flake8](https://flake8.pycqa.org/en/latest/) for that personally, my normal `.flake8` configuration is as follows:

```ini
[flake8]

exclude =
	.git,
	__pycache__,
	.eggs,

max-line-length = 135
indent-size = 1
statistics = True
ignore =
	E124,
	E126,
	E128,
	E131,
	E201,
	E202,
	E203,
	E221,
	E226,
	E241,
	E251,
	E261,
	E262,
	E272,
	E302,
	E303,
	E305,
	E401,
	E402,
	F403,
	F405,
	W191,
	W504
```

Thoughts?

nakengelhardt, 2022-08-04:
Related: https://github.com/YosysHQ/sby/pull/109

lethalbit, 2022-08-04:
Yeah I saw the PR but didn't notice an issue associated with it, forgot to link, it thanks.