YosysHQ/sby #234: `sby` depends on `click` without documenting this or specifying a version constraint

whitequark, 2023-02-27:


nakengelhardt, 2023-02-28:
I did add it to the prerequisites the other day: https://yosyshq.readthedocs.io/projects/sby/en/latest/install.html#prerequisites

whitequark, 2023-02-28:
Ah, I see. It still does not include a version constraint though!

nakengelhardt, 2023-02-28:
Yes, we don't have that for any of the prerequisites - it's not even settled what the minimum python version we still want to support is...

whitequark, 2023-02-28:
Right, so I'm not missing anything then. In YoWASP I'll continue testing on 3.7+ (there are some releases that were installable on 3.6, but ubuntu-latest no longer supports that so I didn't actually test that, and now there's a ~=3.7 constraint). I'll ping you if anything breaks the CI, then.