YosysHQ/sby #282: Verific

AdamKeith1, 2024-06-27:
Getting the error that verific isn't available and curious what it's dependencies were, seemed to work fine in the fifo demo but not working outside of that.

nakengelhardt, 2024-06-28:
Verific is a proprietary frontend (https://www.verific.com) that is only available in the [Tabby CAD Suite](https://www.yosyshq.com/tabby-cad-datasheet). If you are using the OSS CAD Suite, make sure you use only [immediate asserts and assumes](https://yosyshq.readthedocs.io/projects/sby/en/latest/verilog.html#systemverilog-immediate-assertions) supported by the open source frontend, like the 'noverific' version of the fifo demo does.

AdamKeith1, 2024-06-28:
So to my understanding there's no way to use SVA style verification without the Tabby CAD license correct?

jix, 2024-07-02:
Yes, the SVA implementation that is included with yosys depends on the verific frontend only available via Tabby CAD. As far as I know, none of the alternative open source SV frontends that can be used with yosys provide sufficient SVA support and/or yosys integration to allow FV using SVA.

closed 2024-07-02 by jix