YosysHQ/sby #285: Local sva variables assignment is resulting in error.

Humblefool14, 2024-07-22:
SBY 23:10:00 [ff_flat_basic] base: starting process "cd ff_flat_basic/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 23:10:00 [ff_flat_cover] base: starting process "cd ff_flat_cover/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 23:10:00 [ff_flat_basic] base: dv_lib_ff_flat_mem.sv:37: ERROR: Insufficient number of array indices for mem_nxt.
SBY 23:10:00 [ff_flat_cover] base: dv_lib_ff_flat_mem.sv:37: ERROR: Insufficient number of array indices for mem_nxt.

BITDATA and NUMADDR are arbitrary values. 
reg [BITDATA-1:0] mem     [0:NUMADDR-1];
reg [BITDATA-1:0] mem_nxt [0:NUMADDR-1];

always_comb begin
  mem_nxt = mem;

KrystalDelusion, 2025-03-12:
This is expected, see yosys [`tests/svtypes/multirange_subarray_access.ys`](https://github.com/YosysHQ/yosys/blob/main/tests/svtypes/multirange_subarray_access.ys):
```
logger -expect error "Insufficient number of array indices for a." 1
read_verilog -sv <<EOT
module foo;
logic a [6:0][4:0][1:0];
logic b [1:0];

assign a[0][0][0] = 1'b0;
assign a[0][0][1] = 1'b1;
assign b = a[0][0];

endmodule
EOT
```
You're trying to assign the entire array, but the error is telling you that you can only assign one value at a time (meaning you need to iterate over `[0:NUMADDR-1]`).

closed 2025-03-12 by KrystalDelusion

Humblefool14, 2025-03-13:
Thanks, i didn't know this. Will try and work.
How to use the tool better. Is there any documentation which i can go
through to resolving errors ?

Regards,
srikar maturi.

KrystalDelusion, 2025-03-13:
First step would definitely be to ensure the Verilog is error-free by linting with something like [verilator](https://www.veripool.org/verilator/) (there's a brief guide at the end of the [synthesis demo design documentation](https://yosyshq.readthedocs.io/projects/yosys/en/latest/getting_started/example_synth.html#demo-design)).

Then making sure that Yosys is able to handle the design; in your original issue you included `SBY 23:10:00 [ff_flat_basic] base: starting process "cd ff_flat_basic/src; yosys -ql ../model/design.log ../model/design.ys"`, the `ff_flat_basic/model/design.log` can be checked for Yosys errors.  The errors will still be reported by SBY as you saw, but it may help to get more context from the Yosys log directly.  You can also run the `ff_flat_basic/model/design.ys` file yourself, but as the comment at the top says make sure you are in the `ff_flat_basic/src` folder and call `yosys ../model/design.ys`.  If you make any changes to the files in `ff_flat_basic/src` make sure you copy them back to their original location so that you don't lose the changes.  Once the design.ys is working, there will also be design_prep.ys and potentially others depending on what solver(s) you are using.

The main thing is to check the SBY log (e.g. `ff_flat_basic/logfile.txt`) and see exactly where the error was raised.  I don't know if the different steps are listed anywhere, but `base` corresponds to design.ys, `prep` for design_prep.ys, `smt2` for design_smt2.ys etc.  If it reaches `engine_0` or similar then that means it has made it to the solver(s), which I think have their entire log included in the SBY log, otherwise you can check the corresponding `ff_flat_basic/engine_#` folder for logs etc.

Humblefool14, 2025-03-15:
Able to work out using iteration. 
Another issue encountered is module has to be copied within the same file. Why can't we have a different file where another module is defined and instantiated in verified module ?

KrystalDelusion, 2025-03-17:
You should be able to, but you may need to use the defer flag:
> -defer
only read the abstract syntax tree and defer actual compilation
to a later 'hierarchy' command. Useful in cases where the default
parameters of modules yield invalid or not synthesizable code.

`read_verilog -defer <first file> <second file>; hierarchy -top <top module>`

BTW for general questions on how to use Yosys (or SBY) I suggest asking on the community slack https://join.slack.com/t/yosyshq/shared_invite/zt-1aopkns2q-EiQ97BeQDt_pwvE41sGSuA