YosysHQ/sby #287: Importing VHDL packages

AdamKeith1, 2024-07-23:
I have a large vhdl design that uses a few different vhdl packages in multiple locations. I'm having trouble stitching everything together in both the .sby file and my System Verilog testbench. I can't post much any of the vhdl since it's IP. So I'm looking for one of two workarounds...

1. Somehow make a default 'work' library like in QuestaSim/ModelSim. My design has `USE work.axi_pkg.ALL;` style includes and although those package files have been included correctly via the [files] section, they aren't recognized by the design as the 'work' library doesn't exist.

or...

2. Include these vhdl packages in my SV test-bench. I tried to do this via `import axi_pkg::*;` but got the following error 
`starting process "cd fpga_config_bmc/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY  9:07:11 [fpga_config_bmc] base: tb_fpga_config.sv:6: ERROR: syntax error, unexpected TOK_PACKAGESEP, expecting '(' or '['` 
so I'm assuming I'm including incorrectly or need to somehow specify a location to import from. 

**It's important to note that my design and all the packages are appearing the src folder, it's more a matter of figuring out how to make sure that either the test-bench or my design can use the constants in those packages.** 
Or if there's a ghdl way to elaborate a vhdl package that may work too.

whitequark, 2024-07-23:
> 2\. I tried to do this via `import axi_pkg::*;` but got the following error

Only the commercial version of Yosys supports this syntax.

AdamKeith1, 2024-07-23:
Ah, I guess i'll have to find a purely ghdl way to resolve this. Also does the Tabby CAD Suite do evaluation licenses? The SVA support and this vhdl support may make it worth it for us but hard to tell without trying it out.

whitequark, 2024-07-23:
You'd have to ask someone doing sales, I work on the OSS side.

AdamKeith1, 2024-07-23:
Also, is it possible to make subdirectory of design files that would then get generated in the src folder...
ex.
> desing_bmc/src
   design.vhd
   > work
      package1.vhd
      ...
      packagen.vhd

I'm not positive but im thinking if the design is on the same level as this work folder that the `USE work.axi_pkg.ALL;` statements will work. I just am not sure how to generate a folder of design files in the sby file instead of just having them all listed on the same level. Any clue on how to do this? my guess is in the [files] or [scripts] section

AdamKeith1, 2024-07-23:
Clarification on above...

QuestaSim/ModelSim/Vivado all have support for libraries with the default one being 'work'. My vhdl imports packages from this 'work' library. I'm wondering if it's possible to make my own custom library in sby like you can in QuestaSim.

AdamKeith1, 2024-07-24:
Fixed

closed 2024-07-24 by AdamKeith1