YosysHQ/sby #288: Blackbox

AdamKeith1, 2024-07-23:
Is there a means to blackbox modules in sby?

georgerennie, 2024-07-23:
Yes, using the Yosys `blackbox` command in the scripts section. e.g. the following yosys script blackboxes the module bb, so it just exists as a cell with the blackbox attribute set and contents ignored.

```shell
read_verilog <<EOT
module top(input a, output y);
bb b(.a(a), .y(y));
endmodule

module bb(input a, output y);
assign y = !a;
endmodule
EOT

blackbox bb
prep -top top -flatten
show
```

closed 2024-07-23 by georgerennie

AdamKeith1, 2024-07-23:
Do I actually need to include the SV in the [scripts] section or is that just for reference?

georgerennie, 2024-07-23:
Oh thats just for reference. SBY is a wrapper around yosys, and the scripts section is just a set of yosys commands. You can see a list of available yosys commands by starting yosys (`yosys` on the command line) and running `help`.

AdamKeith1, 2024-07-23:
Do you know if this works for vhdl modules as well or just SV?

georgerennie, 2024-07-23:
Yosys makes no distinction internally, a module could have come from either SV or VHDL or something else, it works with a somewhat generic netlist representation.