YosysHQ/sby #300: Generated test banches cannot be used by Xcelium or Iverilog

zputrle, 2024-10-11:
The test benches generated by Sby when an assertion fails cannot be used with the Xcelium or Iverilog simulators.

The problem that I encountered is that, when you try to use the generated test benches with Xcelium and Iverilog, they report that hierarchical name component lookup fails:
```
xmelab: *E,CUVHNF (../rtl/trace7_tb.v,460|59): Hierarchical name component lookup failed at 'UUT.i_byd_axi_mlic.i_dec.\dec_for_loop[0] '.
    UUT.i_axi_mlic.i_dec.\dec_for_loop[0] .i_dec_rd.len_cnt = 8'b00000000; 
```
After discussing this with @nakengelhardt on Yosy's slack, he pointed out that "Yosys doesn't retain enough information about the original structure of the RTL to be able to correctly address the signals that need to be set to the right initial state for the simulation to reproduce the same result."

I'm posting this as an issue so that others looking into using the generated test benches will know that currently they cannot be used.

KrystalDelusion, 2025-02-27:
I think there's a bit of a misnomer here; the generated test benches (e.g. `.../engine_0/trace0_tb.v`) can *only* be used with the corresponding `.../model/design_prep.il`.  You could convert the RTLIL to verilog with `yosys -p 'write_verilog design_prep.v' design_prep.il` and load it into whatever you want.  You can't, however, use the generated testbench with the original source; whether that is loading into Iverilog or Yosys itself, this is what @nakengelhardt was referring to.

If you want to reword this issue as a feature request for using the generated test benches with the original source code (independent of talking about Xcelium/Iverilog) then that is fine, otherwise this issue will be closed.

If you are having an issue with loading the test bench when using the generated `design_prep.v` as I've suggested above, then there might be an issue with the use of `.` in variable names in a flattened design, which would be better served as an issue on the [Yosys repo](https://github.com/YosysHQ/yosys) (though I suspect if that were the case it would've come up by now).

zputrle, 2025-03-12:
Thank you for the information. I have updated the description to note that this is a feature request and kept the initial issue report for context.