YosysHQ/sby #303: ERROR: Can't resolve task name `\assume

goatgate, 2024-10-20:
starting out with formal verification using symbiyosys, don't know where i went wrong
my mux:
module mux(input d,q,sel,
        output o);
assign o = sel?q:d;

always@(*)
begin
assume(sel==1'b1);
assert(q==1'b1);
end
endmodule
`default_nettype wire
~                        
the sby script:
[options]
mode bmc
depth 20
[engines]
smtbmc yices
[script]
read_verilog mux.v
prep -top mux
[files]
mux.v

KrystalDelusion, 2024-10-20:
It might be trying to read it as Verilog, but iirc assume/assert are SystemVerilog; try `read -sv -formal mux.v` or changing the file to `mux.sv`

goatgate, 2024-10-20:
Got that to work, but am facing issues when trying to verify a dff: module dff (
    input clk,
    input reset,
    input d,
    output reg q
);
    always @(posedge clk) begin
        if(reset)
            q<=1'b0;
        else
            q<=d;
    end
    `ifdef FORMAL
    initial begin
        assume(reset);
    end
    property reset_value_check;
    reset |-> ##1(q == 1'b0);
    endproperty
    always @(posedge clk ) begin
        if(reset)
        begin
            assert(q=='b1);
        end
    end
    `endif 
endmodule
`default_nettype wire, the sby script:[options]
mode bmc
depth 1
[engines]
smtbmc yices
[script]
read -sv -formal -DFORMAL dff.v
prep -top dff
[files]
dff.v, 
 the error:SBY 12:02:00 [dff] engine_0: smtbmc yices
SBY 12:02:00 [dff] base: starting process "cd dff/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 12:02:00 [dff] base: dff.v:16: ERROR: syntax error, unexpected TOK_PROPERTY
SBY 12:02:00 [dff] base: finished (returncode=1)
SBY 12:02:00 [dff] base: task failed. ERROR.
SBY 12:02:00 [dff] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 12:02:00 [dff] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 12:02:00 [dff] summary: engine_0 (smtbmc yices) did not return a status
SBY 12:02:00 [dff] summary: engine_0 did not produce any traces
SBY 12:02:00 [dff] DONE (ERROR, rc=16), sorry for the trouble

KrystalDelusion, 2024-10-21:
Unfortunately the open source verilog front end has very limited support for SVA (like the definition of properties). https://symbiyosys.readthedocs.io/en/latest/quickstart.html#concurrent-assertions shows a small example of how you could rewrite a SVA property into supported SV code.

closed 2024-10-22 by goatgate