YosysHQ/sby #307: ERROR: Assert `formal' failed in passes/memory/memory_map.cc:234

CrazybinaryLi, 2024-10-31:
My yosys version is 0.45+112 (git sha1 c8b42b7d4, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3). I utilized the commands "clk2fflogic" and "chformal -lower" within my script to convert $assume and $assert cells into $check cells. However, I encountered the error message “ERROR: Assert `formal' failed in passes/memory/memory_map.cc:234”.

KrystalDelusion, 2025-02-27:
Are you able to share any code to help reproduce this?  As I understand it, `chformal -lower` converts $check cells into $assume and $assert cells, not the other way around.  Secondly, the assertion you're referring to is checking that `memory_map` was called with the `-formal` flag, which is done automatically by SBY for aiger flows, and smtbmc flows with the `--nomem` flag.  Are you trying to call `memory_map` (or `memory`) yourself?

KrystalDelusion, 2025-03-20:
I'm going to close this issue since as far as I can tell this is a bad SBY configuration.  If you disagree then please comment with more information and we can re-open it.

closed 2025-03-20 by KrystalDelusion