YosysHQ/sby #314: allconst fails in weird ways

Topi-ab, 2024-12-16:
This code gives different results, depending which coding style is adopted. Even though all 4 combinations should give equal results.

```vhdl filename="bug_2014_12_16 .vhdl"
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

/*
	Commenting out between
	- t1 & t2
	- a1 & a2
	
	yields to following results:
	
	t1 & a1 enabled => fail
	t1 & a2 enabled => fail
	t2 & a1 enabled => fail
	t2 & a2 enabled => pass
*/

entity bug_2014_12_16 is
	port(
		clk_in: in std_logic
	);
end;

architecture formal of bug_2014_12_16 is
	attribute anyconst: boolean;
	attribute anyseq: boolean;
	attribute allconst: boolean;
	attribute allseq: boolean;
	
	constant LUT_IN_BITS: integer := 3;
	constant LUT_OUT_BITS: integer := 2;
	constant LUT_IN_PERMS: integer := 2**LUT_IN_BITS;
	
	-- t1 =>
	-- type lut_a is array(LUT_IN_PERMS-1 downto 0) of std_logic_vector(LUT_OUT_BITS-1 downto 0);
	
	-- t2 =>
	type lut_a is array(0 to LUT_IN_PERMS-1) of std_logic_vector(LUT_OUT_BITS-1 downto 0);
	
	signal lut_mem: lut_a;
	attribute anyconst of lut_mem: signal is true;
	
	signal lut_in: std_logic_vector(LUT_IN_BITS-1 downto 0);
	attribute allconst of lut_in: signal is true;
	
	signal lut_out: std_logic_vector(LUT_OUT_BITS-1 downto 0);
begin
	process(all)
	begin
		lut_out <= lut_mem(to_integer(unsigned(lut_in)));
	end process;
	
	default clock is rising_edge(clk_in);

	-- a1 =>
	-- a1: assume lut_out = "01";
	
	-- a2 =>
	a2: assume lut_out(1) = '0' and lut_out(0) = '1';
	
	cover {true};
end;
```

```sby filename="bug_2014_12_16 .sby"
[tasks]
cover

[options]
mode cover
depth 1

[engines]
smtbmc --dumpsmt2 --progress --stbv z3

[script]
ghdl --std=08 bug_2014_12_16.vhdl -e bug_2014_12_16
prep -top bug_2014_12_16

[files]
bug_2014_12_16.vhdl
```

KrystalDelusion, 2025-02-27:
Unfortunately the use of PSL means I can't run this without ghdl.  Looking at what this is doing (testing bit ordering) I'm also inclined to believe this is a frontend issue rather than an SBY one; i.e. the problem is that the RTLIL generated by ghdl is order dependent, so I don't think there is anything that can be done on the SBY (or even Yosys) side.  Assuming this *is* specific to allconst, and that array/vector ordering from ghdl *does* normally translate correctly to RTLIL; there may be some special casing because of the attribute in the ghdl/yosys frontend pass that is causing the ordering to be lost?

KrystalDelusion, 2025-02-27:
I will also note that when running
```
read_verilog << EOF
module top ();
    (* allconst *) reg [1:0] dummy;
endmodule
EOF
dump
```
that the loaded design includes the `$allconst` cell that is driving the dummy register.  Again, I'm not sure about how it works with ghdl; but reading the non-PSL part of your vhdl with the Verific frontend just generates
```
  attribute \allconst "true"
  attribute \src "bug_2014_12_16.vhdl:43.9-43.15"
  wire width 3 \lut_in
```
with no corresponding `$allconst` cell.

closed 2025-03-12 by KrystalDelusion