YosysHQ/sby #315: Inquiry on Cover Mode Engine Support and Performance Impact of Converting Cover to Assert in SymbiYosys

SeddonShen, 2025-03-01:
Hi SymbiYosys contributors,

First of all, thank you for your amazing work on this tool! I’ve been using SymbiYosys for formal verification and really appreciate the flexibility it offers. I noticed that BMC mode supports [a wide range of engines](https://symbiyosys.readthedocs.io/en/latest/reference.html#engines-section), which generate corresponding verification files (e.g., SMT-LIB for smtbmc, BTOR2 for btor, and AIG for aiger). This is fantastic! However, I’m curious why cover mode only supports smtbmc and btor engines.

From my understanding, for the BMC process, **cover(conditionA)** seems logically equivalent to **assert(~conditionA)**. So, it feels like supporting AIG-based engines in cover mode should be relatively straightforward. However, this doesn’t seem to be the case. Since I’m particularly interested in using AIG-based solvers for my verification tasks (as SAT solvers perform faster than SMT solvers in my use case), I tried converting my cover assertions into assert statements (e.g., assert(~conditionA)) and running them in BMC mode. But I noticed that the runtime for assert(~conditionA) in BMC mode (using btor btormc) is longer compared to solving cover(conditionA) in cover mode (also using btor btormc).

Could you help me understand:

- What are the potential impacts of converting cover(conditionA) to assert(~conditionA) for verification? Are there specific reasons why AIG-based engines aren’t supported in cover mode?
- If I really want to use AIG file formats for solving cover statements (due to the faster performance of SAT solvers in my tasks), do you have any suggestions or workarounds?

Thanks so much for your time and insights! I’m really excited to hear your thoughts and learn more about how SymbiYosys works under the hood.

Best regards!

georgerennie, 2025-03-01:
The main difference we have with how sby treats covers and assertions is that assertion engines are allowed to exit on the first trace they find, whereas we want cover engines to report as many reachable properties within the bound as they can. None of the AIG engines supported in SBY at the moment support this behaviour, so they aren't enabled for `cover` mode.

If you had multiple statements that you tried to cover with the `assert(~conditionA)`, if using an engine config that doesn't support multiple traces and the assertions are first coverable in different cycles, you wouldn't get traces covering any of the conditions that need longer than the minimum bound to cover.

SeddonShen, 2025-03-01:
Are you saying that if there are multiple ```cover(A) ... cover(B) ... cover(C) ... ``` , the cover engine will try its best to return one or more ```trace.vcd``` files, covering as many of these cover points as possible?

georgerennie, 2025-03-01:
Yes exactly, whereas BMC mode engines are allowed to just return the first trace they find which may not cover them all

SeddonShen, 2025-03-02:
OK, I got it! Thx!!!

closed 2025-03-02 by SeddonShen