YosysHQ/sby #316: How to optimize BMC workflow and use alternative SMT2 solvers like bitwuzla?

SeddonShen, 2025-03-15:
Hello,

I have two questions regarding the BMC workflow when using the `yosys-smtbmc` tool:

## Question 1: AIGER Witness Conversion
When using the AIGER engine, I noticed that the AIGER witness needs to be converted to a Yosys witness (`trace_aiw.yw`) using `yosys-witness`. This converted witness is then passed to `yosys-smtbmc` to re-run BMC. This process seems quite time-consuming.

Is there a more efficient way to directly convert an AIGER witness into a VCD waveform without going through the intermediate Yosys witness format?

## Question 2: Using Alternative SMT2 Solvers
In the `yosys-smtbmc` workflow, how can I use an alternative SMT2 solver (e.g., `bitwuzla`) instead of the default `yices` solver?

Specifically, I would like to configure the solver directly in the `.sby` file, rather than modifying the default engine in `sby_mode_bmc.py`. Is this possible, and if so, could you provide guidance on how to do it?

For reference, here is the relevant log output from the current workflow:

```log
...
SBY 17:49:04 [./cover_test] engine_0: Status returned by engine: FAIL
SBY 17:49:04 [./cover_test] smt2: starting process "cd ./cover_test/model; yosys -ql design_smt2.log design_smt2.ys"
SBY 17:49:04 [./cover_test] engine_0: starting process "cd ./cover_test; yosys-witness aiw2yw engine_0/trace.aiw model/design_aiger.ywa engine_0/trace_aiw.yw"
SBY 17:49:04 [./cover_test] engine_0: Converting AIGER witness trace 'engine_0/trace.aiw' to Yosys witness trace 'engine_0/trace_aiw.yw'...
SBY 17:49:04 [./cover_test] engine_0: Using Yosys witness AIGER map file 'model/design_aiger.ywa'
## Here SBY transforms the aiger witness to yosys witness
SBY 17:49:04 [./cover_test] engine_0: Converted 39 time steps.
SBY 17:49:04 [./cover_test] engine_0: finished (returncode=0)
SBY 17:49:05 [./cover_test] smt2: finished (returncode=0)
SBY 17:49:05 [./cover_test] engine_0: starting process "cd ./cover_test; yosys-smtbmc -s yices --noprogress --append 0 --dump-vcd engine_0/trace.vcd --dump-yw engine_0/trace.yw --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc --yw engine_0/trace_aiw.yw model/design_smt2.smt2"
## Here SBY use yosys-smtbmc(yices) to generate vcd waveform
...
```

And here is the relevant snippet from `sby_mode_bmc.py`:
```python
# sby_mode_bmc.py
import re, os, getopt, click
from sby_core import SbyProc

def run(task):
    task.handle_int_option("depth", 20)
    task.handle_str_option("aigsmt", "yices") # here default to use yices, how to change yices to other solver like bitwuzla?

    for engine_idx, engine in task.engine_list():
        task.log(f"{click.style(f'engine_{engine_idx}', fg='magenta')}: {' '.join(engine)}")
        task.makedirs(f"{task.workdir}/engine_{engine_idx}")
...
```

Thank you for your time and assistance! Looking forward to your response.

Best regards

georgerennie, 2025-03-15:
## Question 1

Do you have an example where the trace reconstruction with smtbmc is taking a significant proportion of runtime? Although we are using smtbmc, we aren't running a full BMC check. In particular, we solve only at the depth of the trace found by the other engine, and constrain all initial variables/inputs to take the values from the trace. This means that the smtbmc check is most of the time just doing constant propagation (which is the same as simulation and fast). The main reason it uses smtbmc is because we have the most infrastructure built on smtbmc and it makes it easiest to uniformly generate counterexample files no matter which engine originally found them.

## Question 2

Symbiyosys allows you to choose what SMT solver is used with the smtbmc engine with a line in the `[engines]` section such as `smtbmc z3`. See the documentation at https://symbiyosys.readthedocs.io/en/latest/reference.html#engines-section.

Note that current mainline bitwuzla is not supported in smtbmc yet (see https://github.com/YosysHQ/oss-cad-suite-build/issues/87 for more. looking at the comments in that it seems the performance issues may have been addressed and we should add support for it but we haven't yet).

SeddonShen, 2025-03-15:
OK, I have no problem in Question1, But I still have question in Question2, I mean in the process of using .yw for smtbmc to generate .vcd file, How to select other slovers like booletctor instead of yices.

For example, now the sby of engine I use is aiger aigbmc
```sby
[engines]
aiger aigbmc
```
It will re run `yosys-smtbmc -s yices --noprogress --append 0 --dump-vcd engine_0/trace.vcd --dump-yw engine_0/trace.yw --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc --yw engine_0/trace_aiw.yw model/design_smt2.smt2"` to generate the vcd file.

But if i use the below sby:

```sby
[engines]
aiger aigbmc
smtbmc z3
```

For engine_0, it will use aigbmc fow a .yw file, and then use `yosys-smtbmc -s yices` to generate vcd file
For engine_1, it will use smtbmc to do word-level model checking...

I mena for engine_0, I don't want use  '-s yices' for yosys-smtbmc, I want use other solvers like z3, boolector.....

By the way, I download the latest oss-cade-suite release, in it's bin actually have bitwuzla program, and I can use
```
[engines]
smtbmc bitwuzla
```
in sby file, I don't what you mean about smtbmc not support bitwuzla?

georgerennie, 2025-03-15:
You can't change which solver is used to recreate the counterexample, but this shouldn't be an issue in practice. As I said, when smtbmc is called after another solver to recreate the counterexample, it isn't doing full bounded model checking, its effectively just using the SMT solver as a simulator so this check should be really fast (if you have some example where it isn't please share it). Therefore although for general BMC there will be a difference in performance between yices and bitwuzla, for this task it shouldn't make much difference (and in my experiments yices often tends to be faster at this type of preprocessing anyway)

When I say smtbmc doesn't support bitwuzla, I just mean it doesn't support the latest version. If you are using oss-cad-suite then you should be fine as the version of bitwuzla included in oss-cad-suite is the version supported by smtbmc.

SeddonShen, 2025-03-15:
OK, I got it , very thanks for your reply!
Thx again for this awesome work, sby really help me a lot!