YosysHQ/sby #318: VHDL assert severity is ignored

JakubFranek, 2025-03-20:
Hello,

first of all I have no idea if this is the correct place to ask this question, please advise me where to post it instead if this is not the correct place.

I have managed to setup SymbiYosys with GHDL and it runs fine. 

**It seems that any failing assert causes the formal verification to fail, regardless of specified severity of the assert.**

For example, suppose I have an assert with severity `note` like this:

```
assert_note: assert (i_clk = '1')
        report "input clock is high"
        severity note;
```

GHDL produces the following reports as expected:

```
D:\Electronics\FPGA\Nexys-A7-Lab\source\clock_enable_generator\clock_enable_generator.vhd:63:5:@20ns:(assertion note): input clock is high
D:\Electronics\FPGA\Nexys-A7-Lab\source\clock_enable_generator\clock_enable_generator.vhd:63:5:@40ns:(assertion note): input clock is high
...
```

and SymbiYosys verification fails because of failing assert.

```
...
SBY 22:35:56 [simulation/sby_bmc] summary: engine_0 (smtbmc z3) returned FAIL
SBY 22:35:56 [simulation/sby_bmc] summary: counterexample trace: simulation/sby_bmc/engine_0/trace.vcd
SBY 22:35:56 [simulation/sby_bmc] summary:   failed assertion clock_enable_generator.assert_note at  in step 0
...
```

Other verification tools like VUnit by default ignore low severities and only fail tests if an assert with severity `error` or `failure` fails.

Is there a way to make SymbiYosys ignore asserts with severity less than a user-specified level? Thanks.

Kind regards
Jakub

KrystalDelusion, 2025-03-20:
Yosys `$assert` cells don't contain severity levels, so at the moment no there is no way to do that automatically.  If you label your assertions with their severity level, like you have with the `assert_note` here you could split them out into a separate test, adding a `chformal -assert [-remove|-assert2assume] *_note` at the end of your SBY `[script]` section.  That would either remove all of the `$assert` cells with a name ending in `_note` to, or convert them to assumptions.  Another step further would be something like
```
[engine]
note: smtbmc bitwuzla --keep-going
~note: smtbmc bitwuzla

[script]
..
~note: chformal -assert -remove *_note
```
where you remove the assertions if you're not running the `note` tag, but if you are running the `note` tag then you run `smtbmc` with the `--keep-going` flag so that it doesn't stop if/when the notes fail.

KrystalDelusion, 2025-03-21:
Ah sorry, the leading `*` doesn't match modules.  Testing with the [fifo example included in SBY](https://github.com/YosysHQ/sby/tree/main/docs/examples/fifo) the assertion cells are named e.g. `fifo/a_raddr`, and `*addr` doesn't match anything, but `fifo/*addr` or `*/*addr` do match both `fifo/a_raddr` and `fifo/a_waddr`.

My suggestion would be to add a `select -write asserts.sel t:$assert` after the `prep` command, which will write the names of all the assert cells to `simulation/sby_bmc/model/asserts.sel`.  Unfortunately I don't have GHDL so can't test your provided example, and trying it with Verific tells me the assert cell is called `sby_severity_test/$verific$i8$sby_severity_test.vhd:31$13` instead of the provided name...

KrystalDelusion, 2025-03-21:
Oh you included the `design.il` right there, hah.  So in your case the cell is called `sby_severity_test/assert_note`, so the `*/*_note` would work fine.

JakubFranek, 2025-03-22:
Hello @KrystalDelusion,
thanks a lot, the `*/*_note` pattern works. I will close this issue now.

closed 2025-03-22 by JakubFranek